libopencm3/libopencm3 #1002: I2C on stm32L4 (nucleo L476)

Salamandar, 2018-12-23:
I'm trying to make the i2c work on stm32l4, but I can't.

Here is my full i2c code : 

	gpio_mode_setup(GPIOB, GPIO_MODE_AF, GPIO_PUPD_NONE, GPIO8 | GPIO9);
	gpio_set_output_options(GPIOB, GPIO_OTYPE_OD, GPIO_OSPEED_HIGH, GPIO8 | GPIO9);
	gpio_set_af(GPIOB, GPIO_AF4, GPIO8 | GPIO9);

	rcc_periph_clock_enable(RCC_I2C1);
	rcc_periph_clock_enable(RCC_GPIOB);
	// rcc_set_i2c_clock_hsi(I2C1);

	i2c_reset(I2C1);
	i2c_set_speed(I2C1, i2c_speed_sm_100k, 8);
	i2c_peripheral_disable  (I2C1);
	i2c_set_7bit_addr_mode  (I2C1);
	i2c_enable_stretching   (I2C1);
	i2c_enable_analog_filter(I2C1);
	i2c_peripheral_enable   (I2C1);

The commented line was used on the stm32f4 example. It just doesn't exist for L4.
I guess I'm missing some clock setup… What should be done ? (When that'll work, I'll be able to do an example for your libopencm3-examples repo)

JiriBilek, 2018-12-23:
Hi, are you sure you have parameters for i2c_transfer7 function correct?
I am working with f0 and for sending data I am using something like
`i2c_transfer7(I2C1, I2C_DEVICE_ADDR, &a, 1, nullptr, 0);`

Salamandar, 2018-12-23:
Yeah, does not work either. One syntax is write, the other is read.

JiriBilek, 2018-12-23:
Just to be sure, are you aware that in the code you posted, your first argument 0x52 in the function i2c_transfer is nonsense? I think it should be I2C1, i.e. 0x40005400, hopefully.

Salamandar, 2018-12-27:
Uh, yeah, i'm using I2C1. I just mixed my head up with HAL while writing the issue.

ChuckM, 2018-12-27:
You want 
```
gpio_mode_setup(GPIOB, GPIO_MODE_AF, GPIO_PUPD_PULLUP, GPIO8 | GPIO9);
```

karlp, 2018-12-30:
well, even internal pullups aren't always enough.

karlp, 2018-12-30:
but yes, you want something like that clock select line, only poking the (RCC_CCIPR) register bits.  That function doesn't eixst for l4.  You can have a look at the f0 /  f3 miplmeentations of this.  (f4 doens't have this....)

Or, don't set it to be based on hsi, but then change the parameter to the set_speed function to match whatever the default is (pclk iirc)

fivdi, 2019-07-28:
The clock for the GPIOB peripheral also needs to be enabled before calling `gpio_mode_setup`, `gpio_set_output_options` and `gpio_set_af`. In other words, the following line of code
```c
rcc_periph_clock_enable(RCC_GPIOB);
```
should be the first line of code in the sample code posted above rather than the 5th line of code as is currently the case.

MarSik, 2019-08-05:
I think the issue is elsewhere (at least for some chips). I am debugging the same thing on STM32L031 and the Reference manual says (RM0377; Example A.14.6) you should put the first data byte into I2C_DR before triggering the start condition. The libopencm3 implementation does not do that and starts by waiting for TXIS instead. My debugging shows that at least my micro puts the start condition on the line and then ends up stuck in a loop with I2C_ISR = 0x8001 (BUSY + TXE).

MarSik, 2019-08-05:
So, the code does work even though the RM does the DR and start in different order. I had another issue (gpio + bad slave chip). So this is not it.