libopencm3/libopencm3 #1011: USB on STM32F07 not working

cactorium, 2019-01-23:
Hello! I can't seem to get the CDCACM example working on the stm32f070f6p6. It looks like it's failing during enumeration; I get a lot of these messages when it's plugged in:

[990181.810165] usb usb2-port1: unable to enumerate USB device
[990183.076817] usb 2-1: new full-speed USB device number 60 using xhci_hcd
[990183.200147] usb 2-1: device descriptor read/64, error -71
[990183.430177] usb 2-1: device descriptor read/64, error -71
[990183.660101] usb 2-1: new full-speed USB device number 61 using xhci_hcd
[990183.783577] usb 2-1: device descriptor read/64, error -71
[990184.013445] usb 2-1: device descriptor read/64, error -71
[990184.120173] usb usb2-port1: attempt power cycle
[990184.763407] usb 2-1: new full-speed USB device number 62 using xhci_hcd
[990184.763589] usb 2-1: Device not responding to setup address.
[990184.970234] usb 2-1: Device not responding to setup address.
[990185.176793] usb 2-1: device not accepting address 62, error -71
[990185.300141] usb 2-1: new full-speed USB device number 63 using xhci_hcd
[990185.300395] usb 2-1: Device not responding to setup address.
[990185.506950] usb 2-1: Device not responding to setup address.
[990185.713398] usb 2-1: device not accepting address 63, error -71
[990185.713457] usb usb2-port1: unable to enumerate USB device
[990185.850058] usb 2-1: new full-speed USB device number 64 using xhci_hcd

Would you guys have any advice on how to debug? I'm using it on a custom board, but I've ordered the stm32f072bdiscovery to replicate. Code is available here: https://github.com/cactorium/auto-orchestra/blob/master/fw/receiver-adc/src/main.c Schematic is available here: https://github.com/cactorium/auto-orchestra/blob/master/pcb/receiver/receiver.pdf

karlp, 2019-01-23:
Can you adapt the f072 gadget zero tests for your board? (It should be trivial) https://github.com/libopencm3/libopencm3/blob/master/tests/gadget-zero/main-stm32f072disco.c

cactorium, 2019-01-23:
Agh, it won't fit apparently

I'm trying it out on this branch: https://github.com/cactorium/auto-orchestra/tree/gadget0-test/fw/receiver-adc/src It's currently 4k bytes over, using -Os and -Wl,--gc-sections : 

[kelvin@batmanbatman src]$ make
  LD      receiver.elf
arm-none-eabi-gcc main.o -Wl,--start-group -lc -lgcc -lnosys -Wl,--end-group -Wl,--gc-sections -lopencm3_stm32f0 -static -nostartfiles -L../libopencm3/lib -Tgenerated.stm32f070f6p6.ld -mcpu=cortex-m0 -mthumb -msoft-float  -o receiver.elf
/usr/lib/gcc/arm-none-eabi/8.2.0/../../../../arm-none-eabi/bin/ld: receiver.elf section `.text' will not fit in region `rom'
/usr/lib/gcc/arm-none-eabi/8.2.0/../../../../arm-none-eabi/bin/ld: region `rom' overflowed by 4328 bytes
collect2: error: ld returned 1 exit status
make: *** [../libopencm3/mk/gcc-rules.mk:44: receiver.elf] Error 1

cactorium, 2019-01-23:
On the bright side I can rule out hardware issues; shorting the BOOT0 pin to start bootloader seems to work fine; the STM DFU bootloader shows up fine in lsusb and dmesg

cactorium, 2019-01-23:
Commenting out ER_DEBUG got it to fit, but it still has the same error messages as the first version

cactorium, 2019-01-24:
I figured it out, it looks like basically the only issue is that RCC_SYSCFG_COMP isn't enabled, which I guess the USB module needs, and the PA10/PA11 remapping seems to be important too (set in SYSCFG_CFGR1). I'm using the stm32f070f6p6, which is the lowest pin count of the family, so the remapping isn't necessary on any other STM32F070xx chip

I dunno if this requires any changes to the library, but I guess this issue might make for good documentation for anyone who stumbles across this

closed 2019-01-24 by cactorium

karlp, 2019-01-24:
Yes, the SYSCFG clock needs to be enabled to use the SYSCFG block, and you need the SYSCFG remap for the small packages yes.  Not sure where we would document this unfortunately.  You don't need the SYSCFG block for the USB itself, (or it would be needed for other parts) just for the pin remapping, which is specific to your package.  Glad you got it sorted!