libopencm3/libopencm3 #1012: Modifying F103 example for F303RE Discovery

danbates2, 2019-01-24:
Hi, I'm trying to modify the '[jeelabs embello](https://github.com/jeelabs/embello/tree/master/explore/1649-f103)' f103 examples for the f303re discovery board.
I've got as far as modifying the makefile architecture flags, linker files, and GPIO definition in the blink.c file example but am getting compilation errors.
Any thoughts?
I'm getting:
`........ uses VFP register arguments, blink.elf does not`

karlp, 2019-01-24:
you're mixing hard/soft float compilations, or mixing cm3 with cm4f compiled pieces, but you should be filing bugs on the project you're modifying, not here.

If you're j ust trying to use f3, perhaps try examples deliberately for f3? https://github.com/libopencm3/libopencm3-examples/tree/master/examples/stm32/f3/stm32f3-discovery/usb_cdcacm

closed 2019-01-24 by karlp

danbates2, 2019-01-24:
Is there support for the nucleo board?

karlp, 2019-01-24:
If you just want blinkies for any board, your best bet is https://github.com/libopencm3/libopencm3-miniblink Otherwise, glad you got it working :)

danbates2, 2019-01-24:
Where can we continue discussion on using cm3?
My email's
dgbates
[ at ] 
mail dot uk