libopencm3/libopencm3 #1019: Genlink generating a blank link script

bruabas, 2019-02-04:
Hi All,
I am using the master version of libopencm3, targeting to STM32F042C4.
In my previous version the linker script file generates correctly (generated.stm32f042c4.ld), but recently it became a blank file and the link fails. If I copy the file I generated with the old version over this blank it works as expected.
I see the lib/ directory only contains .a files and no .ld files, while the previous version had both files.
The library built itself correctly, with no errors.
Did someone experienced something like that as well?

karlp, 2019-02-04:
can you share your project anywhere? or more details about your OS or environment or anything?

bruabas, 2019-02-05:
Sure. I'm using Linux (Debian 9) with official ARM gcc.
The makefile is a pretty canonical one, like this:

```makefile
DEVICE          = stm32f042c4
OPENCM3_DIR     = ~/lib//libopencm3

OBJS            += main.o threads.o list.o chronos.o serial.o delay.o i2c.o accel.o spi.o

CFLAGS          += -Os -ggdb3
CPPFLAGS        += -MD
LDFLAGS         += -static -nostartfiles
LDLIBS          += -Wl,--start-group -lc -lgcc -lnosys -Wl,--end-group

include $(OPENCM3_DIR)/mk/genlink-config.mk
include $(OPENCM3_DIR)/mk/gcc-config.mk

.PHONY: clean all flash

all: binary.elf binary.hex binary.bin

clean:
	$(Q)$(RM) -rf binary.* *.o

include $(OPENCM3_DIR)/mk/genlink-rules.mk
include $(OPENCM3_DIR)/mk/gcc-rules.mk

flash: binary.bin
	st-flash write binary.bin 0x8000000
```
The working version is commit 3af05fb8626889c839469ac1fa505587e2ea6cfa, it generates a complete generated.stm32f042c4.ld; in the other case, this file is blank.
I believe there is something in the installation, because the sample ld files are missing in the /lib dir.
I could not find anything strange so far.

karlp, 2019-02-05:
that working version is very recent.  the only thing touching linker scripts and make files since then is 9551da6b which just added an extra device.  Please try again in a clean checkout?  I would concur that you have somethign odd in your install?

bruabas, 2019-02-13:
I tested with another computer and another distribution and it worked normally.
It seems there is something odd about my installation or with Debian 9.
I will test more, but lets close the issue anyway.
If someone happens to reproduce it, please open again...

closed 2019-02-13 by bruabas