libopencm3/libopencm3 #1026: Infinite loop in i2c_transfer7

kajusK, 2019-02-20:
Hi guys,
is there any specific reason why there's an infinite loop in i2c_transfer7 (file i2c_common_v2.c)?

while (i2c_nack(i2c)); /* FIXME Some error */

It's a really dumb idea to go into the endless loop when i2c device fails to respond. One dead i2c device will take the whole system down... Wouldn't it be better to just return false or something?

karlp, 2019-02-21:
lots of things will do infinite loop.  Feel free to suggest a patch for this if you have concrete changes.  Using a watchdog is also a great idea.

closed 2019-02-21 by karlp

karlp, 2019-02-21:
if you're having a hard time understanding, try asking yourself what your proposal even is.  you even said yourself, "return false, or something"  So... what  _is_ the right thing to do here?  What should the API look like?

kajusK, 2019-02-22:
Please don't take the following text personal, but I would be glad if you read it whole and thing about that for a while.

Well, infinite loop in the library is nonsense, it's broken by desing. It must be up to a user to decide what to do when something goes wrong. Just imagine that the mcu controlls e.g. a drone and uses i2c to talk to something like accelerometer, gyroscope and pressure sensor. Accelerometer dies, therefore no ack arrives and the drone hits the earth/someone/someones property. Watchdog is pointless as the i2c code will be called after boot again, i2c stays in while loop, watchdog reboots the mcu, i2c stays in the while loop...

There are very few places where the infinite loop makes sense in the low level hardware abstraction library - like unhandled exception, stack overflow... The rest should be handled by user. I don't think the i2c nack is the best place for while loop as nack can happen for thousands of reasons (i2c device missing, busy processing the data,...) and user should expect and handle these. Killing the whole system/thread just for a single NACK is a horrible mistake that leas to dead device without a way to ressurect with some error msg or something. It should be up to a user to decide what to do.

Yes, nobody forces me to use that function, I could reimplement it myself, but does it make a sense to use the library at all if I have to rewrite function for multiple peripherals?

Well, what should the api look like? just change the function to return bool and if nack happens, return false. I can a pull request if you want, I'm just not sure what's the opinion of the core developers on api changes. But if this is not a rare exception, it should be brought into more public discussion and general change to error handling should be considered.

Thank you for your time.
Jakub

dzarda, 2021-09-30:
@kajusK I strongly agree with your rationale about not freezing in an infinite loop on bus faults. Returning a result would be ok, but not required. It's usually good enough to handle i2c read results from device-specific register values being returned.

I2C is a beast to handle especially on STMs. There's corner cases after corner cases. But changes are wanted nevertheless.