libopencm3/libopencm3 #1028: RTC for STM32F4 for not working

prashant87, 2019-03-13:
void rtc_firstInit(void)
{
        RCC_APB1ENR |= RCC_APB1ENR_PWREN;
//      printf("PWR enable\n");

        PWR_CR |= PWR_CR_DBP;
        printf("PWR_CR enable\n");

        /* cause a backup domain reset to select the clock source for RTC */
        RCC_BDCR |= RCC_BDCR_BDRST;
        RCC_BDCR &= ~RCC_BDCR_BDRST;
        printf("RCC_BDCR enable\n");

        /* select HSE as clock source for RTC*/
        RCC_BDCR |= RCC_BDCR_RTCSEL_HSE;
        printf("RCC_BDCR HSE clock source selected\n");

        /* enable RTC */
        RCC_BDCR |= RCC_BDCR_RTCEN;
        printf("enable RCC_BDCR by RECEN\n");

        /* enable access to RTC */
        RTC_WPR = (0xCA);
        RTC_WPR = (0x53);
        printf("enable access to RTC\n");
        printf("RTC_ISR: %x\n", RTC_ISR);
        printf("RTC_ISR_INT: %x\n", RTC_ISR_INIT);
        RTC_ISR |= RTC_ISR_INIT;
        printf("RTC_ISR: %x\n", RTC_ISR);
        uint32_t reg_temp = RTC_ISR & RTC_ISR_INITF;
        printf("reg_temp: %x\n", reg_temp);

        //while ((RTC_ISR & RTC_ISR_INITF) == 0); //need to check
        printf("RTC_ISR_INITF\n");

//      RTC_PRER = (99 << RTC_PRER_PREDIV_A_SHIFT); /* asynch prescaler to 100 *
/
//      RTC_PRER |= (9999 << RTC_PRER_PREDIV_S_SHIFT);
                uint32_t sync = 124;
                uint32_t async = 7999;
                rtc_set_prescaler(sync,async);

        /* set date and time */
        dateTime_t time;
        time.dayOfWeek = 4;

        time.days = 01;
        time.months = 01;
        time.year =  87;

        time.hours= 03;
        time.minutes = 03;
        time.seconds = 10;

        rtc_setTimeAndDate(&time);

        RTC_ISR &= ~RTC_ISR_INIT;
}

Still RTC for STM32F407- discovery board is not working . Can someone point me problem

prashant87, 2019-12-17:
Any One?

FloFerrier, 2020-04-21:
Look on the pull request, you can find RTC calendar functions for STM32F4.

julian1, 2020-07-31:
> Still RTC for STM32F407- discovery board is not working . Can someone point me problem

Which board?

Official discovery board has provision for 32.768 crystal. 

But if you zoom in on the board picture, or schematic p28, you see crystal X3 and caps are 'not fitted'.

https://www.stmicroelectronics.com.cn/resource/en/user_manual/dm00039084-discovery-kit-with-stm32f407vg-mcu-stmicroelectronics.pdf

But I am suspicious about why the inexpensive part is not populated.