libopencm3/libopencm3 #1029: STM32F103VCT6 hangs when running from RAM in gpio_set_mode

spacerace, 2019-03-18:
Hi, 
I am trying to execute code in RAM. I have changed the linkerscript to this:

`MEMORY
{
	/* orignal linker settings
	 * rom (rx) : ORIGIN = 0x08000000, LENGTH = 256K
	 * ram (rwx) : ORIGIN = 0x20000000, LENGTH = 48K
     */

    rom (rx) :  ORIGIN = 0x20000000, LENGTH = 32K
	ram (rwx) : ORIGIN = 0x20008000, LENGTH = 16K

}`

This compiles fine and I start st-util and gdb. I load the elf file to 0x20000000 and continue in gdb (execute at entry). Now the board hangs, does nothing.
I stepped through the code and it turns out, it is always hanging at the same line. Here's the gdb screen-log:

`
(gdb) load
Loading section .text, size 0x3c8 lma 0x20000000
Start address 0x200002d4, load size 968
Transfer rate: 10 KB/sec, 968 bytes/write.
(gdb) break main
Breakpoint 1 at 0x20000186: file src/main.c, line 32.
(gdb) continue
Continuing.

Breakpoint 1, main () at src/main.c:32
32	    rcc_periph_clock_enable(RCC_GPIOB);
(gdb) step
rcc_periph_clock_enable (clken=clken@entry=RCC_GPIOB) at ../common/rcc_common_all.c:123
123		_RCC_REG(clken) |= _RCC_BIT(clken);
(gdb) step
main () at src/main.c:33
33	    rcc_periph_clock_enable(RCC_GPIOC);
(gdb) step
rcc_periph_clock_enable (clken=clken@entry=RCC_GPIOC) at ../common/rcc_common_all.c:123
123		_RCC_REG(clken) |= _RCC_BIT(clken);
(gdb) step
main () at src/main.c:36
36	    gpio_set_mode(GPIOB, GPIO_MODE_OUTPUT_2_MHZ, GPIO_CNF_OUTPUT_PUSHPULL, GPIO0);
(gdb) step
gpio_set_mode (gpioport=gpioport@entry=1073810432, mode=mode@entry=2 '\002', cnf=cnf@entry=0 '\000', gpios=gpios@entry=1) at gpio.c:104
104		crl = GPIO_CRL(gpioport);
(gdb) step
105		crh = GPIO_CRH(gpioport);
(gdb) step
110			if (!((1 << i) & gpios)) {
(gdb) step
115			offset = (i < 8) ? (i * 4) : ((i - 8) * 4);
(gdb) step
			tmp32 = (i < 8) ? crl : crh;
(gdb) step
121			tmp32 &= ~(0xf << offset);	/* Clear the bits first. */
(gdb) step
122			tmp32 |= (mode << offset) | (cnf << (offset + 2));
(gdb) step
125			crl = (i < 8) ? tmp32 : crl;
(gdb) step
110			if (!((1 << i) & gpios)) {
(gdb) step
110			if (!((1 << i) & gpios)) {
(gdb) step
110			if (!((1 << i) & gpios)) {
(gdb) step
110			if (!((1 << i) & gpios)) {
(gdb) step
110			if (!((1 << i) & gpios)) {
(gdb) step
110			if (!((1 << i) & gpios)) {
(gdb) step
110			if (!((1 << i) & gpios)) {
(gdb) step
110			if (!((1 << i) & gpios)) {
(gdb) `

I hope someone of you has a clue about this...
If it is not going to work, I will try with stdperiph, but I would love libopencm3....

Thanks,
Nils

karlp, 2019-03-20:
It's probably nothing to do with the gpios, but with how your code is running from ram.  Does it work from flash as you expected?  I seem to recall there being other steps than just setting the addresses to RAM to make it run from ram properly.

spacerace, 2019-03-21:
It works properly from flash.

Here it is described with a F4: https://github.com/libopencm3/libopencm3/wiki/Run-From-RAM
It says to use BOOT0 and BOOT1 pins, but in my case this is not neccessary as I use gdb to set PC to it's address.
Also if I do a reset I want the internal flash's firmware to be booted. I have a working state of the firmware in flash, connect a debugger, run some code from RAM, reset brings back working firmware.
If I tie BOOT0+1 to SRAM boot, my firmware will not be run.

karlp, 2019-03-21:
Ok, but hav eyou checked the RM on running from ram? I suspect there are other things you need to be doing.  It's certainly not a problem in the gpio-set_mode() function, there's no way that would care, it's likely just that your stack might not work, and that was the first time you got to a return or something.

ChuckM, 2019-03-21:
@spacerace the BOOT0 and BOOT1 pins do more than you expect. They change where the chip is expecting to see the interrupt vector table and the top-of-stack address.  You will need to program two things into flash, the reset vector and the top of stack.  Unless you can get the chip to remap those things to address 0, you'll need to change the vector.c file as well to account for the non-standard location of the interrupt vector table. On the F4 you can reprogram it with the SCR register, not sure about the chip you are using.

ChuckM, 2019-03-21:
According to the [Reference Manual](https://www.st.com/content/ccc/resource/technical/document/reference_manual/59/b9/ba/7f/11/af/43/d5/CD00171190.pdf/files/CD00171190.pdf/jcr:content/translations/en.CD00171190.pdf)  You need BOOT0 and BOOT1 to both be at a 'high' level in order to run from RAM by default.

You can of course program a small 'trampoline' in FLASH which basically calls into RAM after setting things up.

a-v-s, 2019-04-01:
Are you sure that linker file is correct.  Your rom and ram overlap, wouldn't this cause your code being overwritten? Try defining a RAM and ROM area that don't overlap.

karlp, 2019-04-02:
32k is 0x800? where's theh overlap?

a-v-s, 2019-04-03:
My mistake, I seem to have missed the 8 last night.

karlp, 2020-01-04:
Did this ever progress further?  I don't have much experience running from ram, and I can't find many people who have tried either.  I've successfully run _functions_ from ram without any problems, but not this specific setup. (I'm not sure I understand the motivation either, but hey, it _should_ work)  I'd still be happy to make changes if there's something we're not doing right here.

karlp, 2020-10-07:
Closing as no further info.  Still interested in fixing, but needs more to do anything meaningful

closed 2020-10-07 by karlp