libopencm3/libopencm3 #1049: File not found nvic.h: libopencm3/stm32/f1/nvic.h

karthikprabhu17, 2019-05-19:
I was trying to build the liopencm3 because its a depedency for [open-ec](https://github.com/rgwan/open-ec) and encountered this file not found error: nvc.h 

Platform:
ubuntu 18.04

Command run:
make

toolchain used
gcc-arm-none-eabi-8-2018-q4-major

Error:
`  CC      usb_private.c
  CC      ec.c
In file included from /home/karthikp/libopencm3/include/libopencm3/cm3/nvic.h:133:0,
                 from ec.c:24:
/home/karthikp/libopencm3/include/libopencm3/dispatch/nvic.h:8:39: fatal error: libopencm3/stm32/f1/nvic.h: No such file or directory
 # include <libopencm3/stm32/f1/nvic.h>
                                       ^
compilation terminated.
../libopencm3.rules.mk:166: recipe for target 'ec.o' failed
make: *** [ec.o] Error 1`

ChuckM, 2019-05-19:
If you haven't built libopencm3 there won't be an nvic.h file (it is generated dynamically). If you are using an old copy of the respository, the header generation will likely fail (it has gone through some iterations). 

Use git to clone a copy of the repository, go to the top level and type 'make'. Once the library is built you can point at it from your application's makefile to link it in.

karthikprabhu17, 2019-05-19:
@ChuckM : Thanks for explaining.. I was able to build libopencm3 .

closed 2019-05-19 by karthikprabhu17

Ha644, 2023-06-25:
can anyone help please.
I found this error trying to flash dps3005
****openocd -f interface/stlink-v2.cfg -f target/stm32f1x.cfg****

Open On-Chip Debugger 0.12.0+dev-01249-g63f4e7c72 (2023-06-14-20:02)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
WARNING: interface/stlink-v2.cfg is deprecated, please switch to interface/stlink.cfg
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Error: open failed