libopencm3/libopencm3 #1057: External interrupt on stm32f4 not working

Prat-123, 2019-05-29:
static void button_setup(void)
{
        /* Enable GPIOA clock. */
        //rcc_periph_clock_enable(RCC_GPIOA); 
       gpio_mode_setup(GPIOE, GPIO_MODE_INPUT, GPIO_PUPD_NONE, GPIO1);
       nvic_enable_irq(NVIC_EXTI1_IRQ);
       exti_select_source(EXTI1,  GPIOE);
       exti_set_trigger(EXTI1, EXTI_TRIGGER_RISING);
       exti_enable_request(EXTI1);
}

//void exti1_isr(void)
{
	exti_reset_request(EXTI1);
	gpio_set(GPIOD, GPIO15);
	printf("ext1\n");
   for (int i = 0; i < 10000000; i++) {	/* Wait a bit. */
				__asm__("nop");
	}
	gpio_clear(GPIOD, GPIO15);
}

i have tried it running on port A it is running absolutely fine, but it is not working on any other port.
please suggest solution.

karlp, 2019-05-29:
have you at least got all the clocks on for all the ports?

Prat-123, 2019-05-29:
Yes.. clock for all ports was on

karlp, 2019-05-29:
Then please attach some more complete code.

Prat-123, 2019-05-31:
#include <libopencm3/stm32/rcc.h>
#include <libopencm3/stm32/gpio.h>
#include <libopencm3/stm32/usart.h>
#include <libopencm3/cm3/nvic.h>
#include <libopencm3/stm32/exti.h>
#include <libopencm3/stm32/syscfg.h>
#include <errno.h>
#include <unistd.h>
#include <stdio.h>

#include "FreeRTOS.h"
#include "task.h"

#define USART_CONSOLE USART6
int _write(int file, char *ptr, int len);

static void clock_setup(void)
{
	rcc_clock_setup_hse_3v3(&rcc_hse_8mhz_3v3[RCC_CLOCK_3V3_168MHZ]);

	/* Enable GPIOD clock for LED & USARTs. */
	rcc_periph_clock_enable(RCC_GPIOD);
	rcc_periph_clock_enable(RCC_GPIOA);
	rcc_periph_clock_enable(RCC_GPIOC);
	rcc_periph_clock_enable(RCC_GPIOE);
	rcc_periph_clock_enable(RCC_GPIOB);

	/* Enable clocks for USART2. */
	rcc_periph_clock_enable(RCC_USART6);
}

static void usart_setup(void)
{
	/* Enable the USART2 interrupt. */
	//nvic_enable_irq(NVIC_USART2_IRQ);

	/* Setup GPIO pins for USART2 transmit. */
	gpio_mode_setup(GPIOC, GPIO_MODE_AF, GPIO_PUPD_NONE, GPIO6);

	/* Setup GPIO pins for USART2 receive. */
	gpio_mode_setup(GPIOC, GPIO_MODE_AF, GPIO_PUPD_NONE, GPIO7);
	gpio_set_output_options(GPIOC, GPIO_OTYPE_OD, GPIO_OSPEED_25MHZ, GPIO7);

	/* Setup USART2 TX and RX pin as alternate function. */
	gpio_set_af(GPIOC, GPIO_AF8, GPIO7);
	gpio_set_af(GPIOC, GPIO_AF8, GPIO6);

	/* Setup USART2 parameters. */
	usart_set_baudrate(USART_CONSOLE, 115200);
	usart_set_databits(USART_CONSOLE, 8);
	usart_set_stopbits(USART_CONSOLE, USART_STOPBITS_1);
	usart_set_mode(USART_CONSOLE, USART_MODE_TX_RX);
	usart_set_parity(USART_CONSOLE, USART_PARITY_NONE);
	usart_set_flow_control(USART_CONSOLE, USART_FLOWCONTROL_NONE);

	/* Finally enable the USART. */
	usart_enable(USART_CONSOLE);
}

int _write(int file, char *ptr, int len)
{
	int i;

	if (file == STDOUT_FILENO || file == STDERR_FILENO) {
		for (i = 0; i < len; i++) {
			if (ptr[i] == '\n') {
				usart_send_blocking(USART_CONSOLE, '\r');
			}
			usart_send_blocking(USART_CONSOLE, ptr[i]);
		}
		return i;
	}
	errno = EIO;
	return -1;
}

static void gpio_setup(void)
{
	/* Setup GPIO pin GPIO12 on GPIO port D for LED. */
	gpio_mode_setup(GPIOD, GPIO_MODE_OUTPUT, GPIO_PUPD_NONE, GPIO12|GPIO13|GPIO15|GPIO14);
}

static void button_setup(void)
{
        /* Enable GPIOA clock. */
        //rcc_periph_clock_enable(RCC_GPIOA);

        /*(Set GPIOE1 to 'input floating'.*/
       gpio_mode_setup(GPIOE, GPIO_MODE_INPUT, GPIO_PUPD_NONE, GPIO1);

       nvic_enable_irq(NVIC_EXTI1_IRQ);
       exti_select_source(EXTI1,  GPIOE);
       exti_set_trigger(EXTI1, EXTI_TRIGGER_RISING);
       exti_enable_request(EXTI1);

}

static void task1(void *args) {
  //int i;

  (void)args;

  for (;;) {
    	printf("D 12\n");
	gpio_toggle(GPIOD, GPIO12);	// LED on/off
	vTaskDelay(800);
		}
  }

static void task3(void *args) {
  //int i;

  (void)args;

  for (;;) {
	  if(gpio_get(GPIOA, GPIO0))
	  {
    	printf("D 1JHHJH 5\n");
	gpio_toggle(GPIOD, GPIO15);	// LED on/off
	vTaskDelay(800);
	  }
		}
  }

TaskHandle_t xHandle;
int main(void)
{
	clock_setup();
	gpio_setup();
	usart_setup();
	printf("hi guyz\n");
	button_setup();

	  xTaskCreate(task1, "LED", 100, NULL, configMAX_PRIORITIES - 1, &xHandle);
	  xTaskCreate(task3, "LED2", 100, NULL, configMAX_PRIORITIES - 1, NULL);
	
	  vTaskStartScheduler();
	  for (;;)
	    ;
	  return 0;
}

uint8_t state =0;  // running state
void exti1_isr(void)
{
	exti_reset_request(EXTI1);
	gpio_set(GPIOD, GPIO14);
	printf("d 14 interrupt\n");
//	vTaskDelay(800);
	if(state==0)
	{
		 vTaskSuspend( xHandle );
		 state = 1;
	}
	else
	{
		vTaskResume( xHandle );
		state = 0;
	}
	for (int i = 0; i < 1000000; i++) {	/* Wait a bit. */
				__asm__("nop");
			}
	gpio_clear(GPIOD, GPIO14);

}

karlp, 2019-06-01:
You need rcc_periph_clock_enable(RCC_SYSCFG) to enable the exti port mappings

closed 2019-06-01 by karlp