libopencm3/libopencm3 #1073: STM32F446 USB Driver Not Functional

vielster, 2019-06-26:
Have attempted getting USB working with present otgfs driver implementation with no success. I have tried with the sample cdcacm example code, as well as custom interrupt driven implementation, and have similar results. 

_Note: This same code (poll'd and interrupt driven) works on an STM32F407 without issue._

I'm running on a Nucleo-F446ZE board. Working through configuration, I was able to get the board to start the enumeration process, but I get repeated USB suspend and ENUMDNE interrupts, and never get callbacks to the registered control callbacks. Linux also reports failed enumeration and retries repeatedly.

Sample output with some debug printouts put in the polling function:
```
USB: HandleIrq
dwc_poll
dwc_poll(ENUMDNE)
USB: HandleIrq
dwc_poll
dwc_poll(SUSPEND)
USB: HandleIrq
dwc_poll
dwc_poll(ENUMDNE)
USB: HandleIrq
dwc_poll
dwc_poll(ENUMDNE)
dwc_poll(SOF)
USB: HandleIrq
dwc_poll
dwc_poll(SUSPEND)
dwc_poll(SOF)
USB: HandleIrq
dwc_poll
dwc_poll(ENUMDNE)
USB: HandleIrq
dwc_poll
dwc_poll(SUSPEND)
dwc_poll(SOF)
USB: HandleIrq
dwc_poll
dwc_poll(ENUMDNE)
USB: HandleIrq
dwc_poll
dwc_poll(SUSPEND)
dwc_poll(SOF)
USB: HandleIrq
dwc_poll
dwc_poll(ENUMDNE)
USB: HandleIrq
dwc_poll
dwc_USB: HandleIrq
dwc_poll
dwc_poll(ENUMDNE)
USB: HandleIrq
dwc_poll
dwc_poll(SUSPEND)
dwc_poll(SOF)
```

I'm able to get rid of the suspend interrupts by adding an update to the usb_f107.c initialization.

```
	OTG_FS_DCTL |= OTG_DCTL_RWUSIG;
```

This gives some improvement, but still does not successfully enumerate.

```
USB: HandleIrq
dwc_poll
dwc_poll(ENuMDNE)
USB: HandleIrq
dwc_poll
dwc_poll(RXFLVL)
dwc_poll(EPSETUP 0x00)
dwc_poll(SOF)
USB: HandleIrq
dwc_poll
dwc_poll(RXFLVL)
dwc_poll(EPCB 0)
dwc_poll(?)
USB: HandleIrq
dwc_poll
USB: HandleIrq
dwc_poll
dwc_poll(RXFLVL)
dwc_poll(EPRW 0x00)
...
```

Here is the sample kernel output while enumeration is being attempted...

```
Jun 26 00:48:38 vielster-glaptop.roam.corp.google.com kernel: [56274.924726] usb 1-1: new full-speed USB device number 17 using xhci_hcd
Jun 26 00:48:43 vielster-glaptop.roam.corp.google.com kernel: [56279.952691] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
Jun 26 00:48:48 vielster-glaptop.roam.corp.google.com kernel: [56285.328772] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
Jun 26 00:48:48 vielster-glaptop.roam.corp.google.com kernel: [56285.540586] usb 1-1: device not accepting address 17, error -62
Jun 26 00:48:49 vielster-glaptop.roam.corp.google.com kernel: [56285.668622] usb 1-1: new full-speed USB device number 18 using xhci_hcd
Jun 26 00:48:54 vielster-glaptop.roam.corp.google.com kernel: [56290.704691] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
Jun 26 00:48:59 vielster-glaptop.roam.corp.google.com kernel: [56296.080812] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
Jun 26 00:48:59 vielster-glaptop.roam.corp.google.com kernel: [56296.288704] usb 1-1: device not accepting address 18, error -62
```

karlp, 2019-06-26:
Repeaitng what I said on IRC,
```
6:34 <karlp> just for fun, do you have a usb 2 port you could try it out with?
06:35 <karlp> when you say "the" usb test sample, which code are you meaning there?
06:36 <karlp> I'd try libopencm3/tests/gadget-zero/ and modify either the f4disco or f429i-disco files depending on whether you're using the usb-fs or usb-hs ports.
06:36 <karlp> I don't have an f446, and the only problems _I remember_ hearing people talk abot on f446 and friends were the changed clocks.
06:37 <karlp> if this is your own hardware, does it work with a cube demo? (just for sanity)
```

Further, how are you printing these debugs?  hopefully via trace, rather than semihosting or something? it's fairly well known that it can interfere if you print out too much.

Stay in the irc channel, you just landed at a time of day when there's not many people around :)

zyp, 2019-06-26:
After @karlp reminded me about my comment on #983, I went back and looked what that was about, and it's probably what you're also seeing.

Try reverting 26ab78a and see if it helps.

OpenUAS, 2019-09-27:
@vielster any feedback on @zyp hint would be welcomed.

vielster, 2019-11-07:
Reverting 26ab78a addressed this issue.

karlp, 2019-11-07:
Unfortunately that doesn't fix our issue, it just gives us more information to try and find a fix that works for all boards.  reopening for that.