libopencm3/libopencm3 #1075: Remove Python 2 dependency (move to Python 3)

Peque, 2019-06-26:
Some distributions, like mine (Fedora 30), do not ship with Python 2 installed by default anymore. Furthermore, Python 2 will be discontinued in 2020.

Would you accept a PR that changes the current `#!/usr/bin/env python` shebang to `#!/usr/bin/env python3`? (with the necessary changes, if required for Python 3 compatibility)

Of course I can install Python 2, but just wondered if it would make sense to consider this change.

karlp, 2019-06-27:
The script is already python3 compatible right?  I was under the impression that "python" was going to be python3, not that "python" would stop being a valid binary, and everyone would need to change their scripts, that sounds wildly intrusive.

karlp, 2019-06-27:
According to https://www.python.org/dev/peps/pep-0394/ we should keep using "python" if we want to support both 2 and 3, and just make sure our script works with both, which I _believe_ it does, but feel free to correct me on that.  If fedora is breaking the PEP by simply having _no_ python executable, then, respectfully, I feel that's a fedora bug. (And thanks for hitting it before me, I'm still on fed29)

karlp, 2019-06-27:
I've just tested this in a python3 venv (python3 -m venv .env3 ; . .env3/bin/activate && make clean && make all) and it runs just fine.  please le tme know if you find any explicitly python2 only code

Peque, 2019-06-27:
> If fedora is breaking the PEP by simply having no python executable, then, respectfully, I feel that's a fedora bug. (And thanks for hitting it before me, I'm still on fed29)

Agree with that. :+1:

It seems this is an [intermediate step for Fedora before making `python` point to `python3`](https://fedoraproject.org/wiki/FinalizingFedoraSwitchtoPython3).

> I've just tested this in a python3 venv (python3 -m venv .env3 ; . .env3/bin/activate && make clean && make all) and it runs just fine. please le tme know if you find any explicitly python2 only code

Yeah, I can confirm that. To be honest I just grepped for shebangs and did not even look at the source code of the scripts. :sweat_smile:

> According to https://www.python.org/dev/peps/pep-0394/ we should keep using "python" if we want to support both 2 and 3 [...]

Accoring to PEP 394, it seems that shebangs should not specify `python`, but rather should specify either `python2` or `python3`, unless deliberately written to be source compatible with both Python 2.x and 3.x, in which case they may continue to use `python` on their shebang line.

So yeah, if they are, and you want them to be, Python 2.x and 3.x compatible, you can leave the shebang as-is.

Allow me to close this for you. Sorry for the noise! :innocent:

closed 2019-06-27 by Peque

sumpfralle, 2020-10-28:
> According to https://www.python.org/dev/peps/pep-0394/ we should keep using "python" if we want to support both 2 and 3, and just make sure our script works with both, which I believe it does, but feel free to correct me on that.

[PEP-394](https://www.python.org/dev/peps/pep-0394/#for-python-script-publishers) allows `#!/usr/bin/env python`, but should only be used in a virtualenv:

> For scripts that are only expected to be run in an activated virtual environment, shebang lines can be written as #!/usr/bin/env python, as this instructs the script to respect the active virtual environment.

I guess, that we do not really expect the users of libopencm3 to run `make` (for compiling libopencm3) in a python virtualenv, or?

PEP-394 further says:

> Scripts targeting both “old systems” and systems without the default python command need to make a compromise and document this situation. Avoiding shebangs (via the console_scripts Entry Points ([9]) or similar means) is the recommended workaround for this problem.

Thus the minimum effort recommended by PEP-394 would be to document, that `make` should be typed in a Python virtualenv. Other approaches could be a new pre-compilation step converting the input python scripts into *locally usable* python scripts with a proper (detected) shebang. But this feels unreasonably cumbersome.

> If fedora is breaking the PEP by simply having no python executable, then, respectfully, I feel that's a fedora bug. (And thanks for hitting it before me, I'm still on fed29)

No, I think, Fedora's removal of `python` does not break PEP-394, but follows its wording perfectly.
See one of the [proposed options](https://www.python.org/dev/peps/pep-0394/#for-python-runtime-distributors):

> Distributors may choose to set the behavior of the python command as follows:
> * ...
> * not provide python command,
> * ...

At least Fedora and Debian (and thus probably all Debian-derivates like Ubuntu, Mint, ...) will *not* ship `/usr/bin/python` at least for one release, in order to give local administrators a clear indication, that they should take a look at certain (old) scripts and check, whether they are ready for switching to another language (python3). *Maybe* these distributions will start providing `/usr/bin/python` in a later release again (after everyone noticed during the previous upgrade, that their local scripts need attention). But this will depend on whether anyone will be interested in `/usr/bin/python` at this point in the future. Since every PEP is recommending to use `python3` (instead of `python`) for years, there may be no need for `/usr/bin/python` in a few years.

Now we are close to the end of 2020:
* python3 is available in every major distribution for many, many years (since ~2010)
    * RHEL5 is probably the last distribution, shipping only python2 (even though there are probably ways to install python3)
    * RHEL5's *Extended Lifecycle Support* will be end next month (November 2020); its general *maintenance* period expired three years ago
* python2 is being removed (or was removed some time ago) from major distributions due to the end-of-life of Python2 (January 2020)

Conclusion:
* `python` is not available on some current distributions and will be missing from *most* distributions during the next years
* `python3` is available in *all* currently supported distributions
    * at least at the end of November 2020 (due to RHEL5, see above)
* sticking to the `python` approach for shebangs would require us to teach users to use a python virtualenv when running `make` or to provide some fancy detection of the python interpreter in libopencm3's Makefiles

Thus I would recommend to switch the shebang of the four related scripts below `scripts/` from `#!/usr/bin/env python` to `#!/usr/bin/env python3`. This will ease the life of the majority of our current and future users, while causing suffering only for users of pre-2010 distributions, which are technically really out-of-date and unsupported.

The future (python3) is here - let us welcome it! :)