libopencm3/libopencm3 #1087: Trouble with Installation

jnestor481, 2019-08-04:
Hello everyone,

First time on Github. Completely new to all this. Engineering student who is trying to learn about microcontrollers. Have to confess I'm not the most computer savy person out there. In fact I think they usually turn into a huge time sink of IT issues. So here I am. 

I bought a book called Beginning STM32 by Warren Gay. I have followed all the instructions as best I could. I downloaded Cygwin, libopencm3, FreeRTOS (10.2.1), and the ARM cross compiler. I am stuck at this command line prompt that he gives:
line 1: $ cd ~/stm32
line 2: make

At first I couldn't use the 'make' command but I fixed that by reinstalling Cygwin with it included.  Then it was an issue with Python, but same fix just reinstalled Cygwin. Now I get the following error message and I'm unsure what to do. I try my best to fix my own problems but hitting my head against a brick wall on this one. 

make -C libopencm3 TARGETS=stm32/f1
make[1]: Entering directory '/cygdrive/c/Users/Joe/AppData/Roaming/SPB_Data/stm32/libopencm3'
  GENHDR  include/libopencm3/stm32/f1/irq.json
/bin/sh: ./scripts/irq2nvic_h: /usr/bin/env: bad interpreter: Permission denied
make[1]: *** [Makefile:61: include/libopencm3/stm32/f1/irq.json.genhdr] Error 126
make[1]: Leaving directory '/cygdrive/c/Users/Joe/AppData/Roaming/SPB_Data/stm32/libopencm3'
make: *** [Makefile:38: libopencm3/lib/libopencm3_stm32f1.a] Error 2

Any help would be hugely appreciated. Thank you.

ASHIT-AXAR, 2019-08-08:
i have trouble installing too.
may someone put a ready to install release file for windows ?

karlp, 2019-08-08:
You're probably running into problems with cygwin, which isn't really all taht nice to use these days.  https://www.msys2.org/ is the "new" "cywgin" style of things on windows, and by all reports is much more reliable and well integrated.

closed 2019-08-08 by karlp