libopencm3/libopencm3 #1089: Add STM32G4 support

jaxxzer, 2019-08-18:
The new [stm32g4](https://www.st.com/en/microcontrollers-microprocessors/stm32g4-series.html) line promises close compatibility with the stm32f3 line.

karlp, 2019-08-18:
sure, PRs welcome, we don't really need a separate issue to track it though.

closed 2019-08-18 by karlp

jaxxzer, 2019-08-19:
I am interested in developing and submitting a PR.

@karlp  please:
- are you aware of anyone working around this already?
- to be sure I approach this according to your expectation, can you outline the starting point or requirements for the PR (ie. can we start with just rcc/gpio support, and add support for peripherals one by one, or do you want all peripherals supported at once? Do you use the SVD files to generate the macros, or are those done by hand normally?)
- if there is a better forum to discuss this, please direct me to it

karlp, 2019-08-19:
no-one else to my knowledge.  Perhaps @grevaillot but I've not seen or heard anything.
Just rcc/gpio is plenty for starters, it's all we normally start with.

We don't normally use SVD files, no.  Have a look at how g0 and l4 were added for recent examples.

irc is easiest for me, we're in #libopencm3 on freenode, but you're always welcome to comment on things here, particularly if it's attached to your PR.  it can be in draft stage, that's fine.

grevaillot, 2019-08-20:
I had a look at the rm/datasheet and planned to push some basic support, but we have no immediate need at work for g4, so nothing here yet for me :)

grevaillot, 2019-08-27:
@jaxxzer despite having plenty of other usefull things to do / merge / fix, i started to have a look on my side - have you started on your side ?

jaxxzer, 2019-08-31:
This openocd patch was recently updated to add support for G4: http://openocd.zylin.com/#/c/4807/

grevaillot, 2019-09-03:
> This openocd patch was recently updated to add support for G4: http://openocd.zylin.com/#/c/4807/

hmm, this one does not reaally work well it seems. Target explodes when trying to poke FLASH register at reset-init - Despite Flash ws/PWR range being ok at reset at first look - it basically clears the DBG_SWEN bit while setting up ACR register. This looks dubious - following RCC code does not look really usefull too.

jaxxzer, 2019-09-03:
Oh, yes I noticed the same after I commented here, then tried. I did forget to follow up.

The g0 support was working before the patch was updated, I didn't check after the update with broken g4.