libopencm3/libopencm3 #1090: would you provide a ready to install release of libopencm3 ?

ASHIT-AXAR, 2019-08-18:
would you please provide a install file of libopencm3 ?
i get trouble installing build environment. not familiar with msys, phyton, ...
i couldn't pull those together.

[image]

$ make
bash: make: command not found

ChuckM, 2019-08-18:
Hi @ASHIT-AXAR , I realize it doesn't help you directly but if you are running Windows 10 the Microsoft supplied Ubuntu Linux (called Windows System Linux) from the Microsoft store builds the library very easily.

That said, what environment do you normally use to compile and link against GCC build binaries? You could install Cygwin with Python and bash and that would probably work for now.

karlp, 2019-08-18:
Setting up your environment is out of the scope of what we can really help with, certainly via issues at least.  msys2 should be fine, but you'll need to install make (via pacman) at least yet.

closed 2019-08-18 by karlp

ASHIT-AXAR, 2019-08-20:
axar@ashit MSYS /C/Users/axar/Downloads/Compressed/libopencm3-master
$ make
'printf' is not recognized as an internal or external command,
operable program or batch file.
Makefile:60: recipe for target 'include/libopencm3/stm32/f0/irq.json.genhdr' failed
make: *** [include/libopencm3/stm32/f0/irq.json.genhdr] Error 1

karlp, 2019-08-20:
you still have something wrong with your msys install then.  I can't help you further with this on windows I'm afraid.