libopencm3/libopencm3 #1091: stm32f071xb timer 1 not counting

W-M-D, 2019-08-22:
Hello i'm trying to get my timer to read encoder input. I have confirmed that there is encoder input on these pins and that the processor can read them normally. What am I doing wrong in the configuration?  Thanks 

    rcc_periph_clock_enable(RCC_TIM1);
    gpio_mode_setup(GPIOE, GPIO_MODE_AF, GPIO_PUPD_NONE, GPIO13);
    gpio_mode_setup(GPIOE, GPIO_MODE_AF, GPIO_PUPD_NONE, GPIO14);
    gpio_set_af(GPIOE,GPIO_AF0,GPIO13);
    gpio_set_af(GPIOE,GPIO_AF0,GPIO14);
    
    TIM_ARR(TIM1) =1024;
    timer_slave_set_mode(TIM1, 0x3); // encoder
    timer_ic_set_input(TIM1, TIM_IC3, TIM_IC_IN_TI1);
    timer_ic_set_input(TIM1, TIM_IC4, TIM_IC_IN_TI2);
    timer_enable_counter(TIM1);

karlp, 2019-08-22:
Have you enabled RCC_GPIOE?

W-M-D, 2019-08-23:
yes

W-M-D, 2019-08-23:
It would appear that EXTI0 is also not seeing the encoder pulses or at least not triggering the interrupt :/

karlp, 2019-08-23:
have you enabled exti, both it's own irq _and_ in the nvic? do you get exti irqs in general?  what does work?

W-M-D, 2019-09-12:
i got the exti working in modem because it has an abandonment handler. It turns out that the interrupt was going to an unhandled channel. I'm a bit confused by     timer_ic_set_input what is the channel of the timer and what's the channel of the pin. It says source but does it mean clock source. ugh

karlp, 2020-01-04:
Is there something that we can work on in the documentation here? I'm not sure I follow about the timer docs and exti and what an "abandonment handler" is?

karlp, 2020-04-09:
Feel free to re-open this if you have any updates.

closed 2020-04-09 by karlp