libopencm3/libopencm3 #1096: 【.gitignore】Need to include some files.

webees, 2019-08-30:
These files should not be ignored：

[image]

https://github.com/libopencm3/libopencm3/blob/master/.gitignore

karlp, 2019-08-31:
Can you elaborate?  The patterns do match a couple of explicitly committed files, but git doesn't ignore committed files. so... not sure what you're getting at?  Try it, edit one of those files and you'll see that git status still correctly shows you the changes to them.  Feel free to re-open if you think there's something more here.

closed 2019-08-31 by karlp

webees, 2019-08-31:
When I submit the libopencm3 library code from the local to git, the git repository will get some files missing.

karlp, 2019-08-31:
please demonstrate further, this is clearly not a problem for anyone else contributing, so I'm completely at a loss to understand your concerns here.