libopencm3/libopencm3 #1099: trouble in compiling

ASHIT-AXAR, 2019-09-06:
hello
i get this error when compiling libopencm3. i did step by step described in windows section install guide but unable to get things right.
i installed 3 prerequisites mentioned there (msys, python27, gcc-arm-none-ebi)
is there anyone who compiled for windows recently? i don't know which step i'm doing wrong.

[image]

ChuckM, 2019-09-06:
Not a supported build environment. If you were on Windows 10 I would suggest you install Microsoft's "Windows System Linux" from the Microsoft store and build it that way.

ASHIT-AXAR, 2019-09-08:
axar@ashit MSYS ~
# export PATH=$PATH:"C:\\Program Files\\Git\\bin:C:\\Program Files (x86)\\GNU Tools ARM Embedded\\8 2019-q3-update\\bin:C:\\Program Files (x86)\\CodeBlocks\\MinGW\\bin:c:\\python27:/usr/local/bin:/usr/bin:/bin"

axar@ashit MSYS ~
# cd "C:\\Users\\axar\\Downloads\\Compressed\\libopencm3-master"

axar@ashit MSYS /c/Users/axar/Downloads/Compressed/libopencm3-master
# make
  GENHDR  include/libopencm3/stm32/f0/irq.json
  GENHDR  include/libopencm3/stm32/f1/irq.json
  GENHDR  include/libopencm3/stm32/f2/irq.json
  GENHDR  include/libopencm3/stm32/f3/irq.json
  GENHDR  include/libopencm3/stm32/f4/irq.json
  GENHDR  include/libopencm3/stm32/f7/irq.json
  GENHDR  include/libopencm3/stm32/l0/irq.json
  GENHDR  include/libopencm3/stm32/l1/irq.json
  GENHDR  include/libopencm3/stm32/l4/irq.json
  GENHDR  include/libopencm3/stm32/g0/irq.json
  GENHDR  include/libopencm3/gd32/f1x0/irq.json
  GENHDR  include/libopencm3/lpc13xx/irq.json
  GENHDR  include/libopencm3/lpc17xx/irq.json
  GENHDR  include/libopencm3/lpc43xx/m4/irq.json
  GENHDR  include/libopencm3/lpc43xx/m0/irq.json
  GENHDR  include/libopencm3/lm3s/irq.json
  GENHDR  include/libopencm3/msp432/e4/irq.json
  GENHDR  include/libopencm3/efm32/tg/irq.json
  GENHDR  include/libopencm3/efm32/g/irq.json
  GENHDR  include/libopencm3/efm32/lg/irq.json
  GENHDR  include/libopencm3/efm32/gg/irq.json
  GENHDR  include/libopencm3/efm32/hg/irq.json
  GENHDR  include/libopencm3/efm32/wg/irq.json
  GENHDR  include/libopencm3/efm32/ezr32wg/irq.json
  GENHDR  include/libopencm3/sam/3a/irq.json
  GENHDR  include/libopencm3/sam/3n/irq.json
  GENHDR  include/libopencm3/sam/3s/irq.json
  GENHDR  include/libopencm3/sam/3u/irq.json
  GENHDR  include/libopencm3/sam/3x/irq.json
  GENHDR  include/libopencm3/sam/4l/irq.json
  GENHDR  include/libopencm3/sam/d/irq.json
  GENHDR  include/libopencm3/vf6xx/irq.json
  GENHDR  include/libopencm3/swm050/irq.json
  BUILD   lib/stm32/f0
/usr/bin/sh: -c: line 0: syntax error near unexpected token `('
/usr/bin/sh: -c: line 0: `C:/Program Files (x86)/CodeBlocks/MinGW/bin/make --directory=lib/stm32/f0 SRCLIBDIR="C:/Users/axar/Downloads/Compressed/libopencm3-master/lib" PREFIX="arm-none-eabi-" || \'
Makefile:69: recipe for target 'lib/stm32/f0' failed
make: *** [lib/stm32/f0] Error 1

ASHIT-AXAR, 2019-09-08:
ok, it looks that will not work in windows. i might install ubuntu someday :sweat_smile: :sweat_smile:

karlp, 2020-01-04:
I suspect this is choking on the (x86) in your program files path but I can't understand why? It's inside make on windows as far as I can tell, and libopencm3 only uses $(MAKE) in our code, so I've no idea where to go next.  @ASHIT-AXAR did you get this resolved?  I might take me some time to track down a windows machine to try this on, but as far as I know, it _should_ work, and we've taken (and will continue to take) patches to make sure things work on windows, so it's _meant_ to work

Jak-o-Shadows, 2020-01-05:
I just tested on my Windows 10 pc. It worked. I suspect the difference is in the MSYS paths somewhere - as I did the non-recommended overwriting of default windows binaries with the MSYS.

Perhaps try:
`export PATH="/C/Program Files (x86)/CodeBlocks/MinGW/bin:/c/python27:/C/Program Files (x86)/GNU Tools ARM Embedded/8 2019-q3-update/bin:/usr/local/bin:/usr/bin:/bin"$PATH:` instead of the command you did?

For the record, my system path is:
1. `C:\Users\jak\.cargo\bin` (Rust stuff)
2. `E:\Tools\arm-none-eabi\8 2019-q1-update\bin`
3. `%USERPROFILE%\AppData\Local\Microsoft\WindowsApps`
4. `E:\Tools\MinGW\msys\1.0\bin`
5. `E:\Tools\Microsoft VS Code\bin`
6. `E:\Tools\PothosSDR\bin`

ASHIT-AXAR, 2020-01-05:
oh, sorry i forgot to close this thread.
no, it's not resolved. i left it for another time, as i'm not in hurry.
later i found there is another compiler "MinGW64_x64"(maybe for newer systems?) which is different than my "MinGW_x64" setup.
i will try that later...

closed 2020-01-05 by ASHIT-AXAR