libopencm3/libopencm3 #1106: stm32f3-rtc: Common head does not exist

h2obrain, 2019-09-25:
I just stumbled over this looking for a rtc implementation suitable for the f7. 
The libopencm3/stm32/common/rtc_common_all.h referenced in [f3/rtc.h](https://github.com/libopencm3/libopencm3/blob/5fbe5c8c477ad97f21a2f3f523f9d422a80fd385/include/libopencm3/stm32/f3/rtc.h#L39) does not seem to exist. Is it the same as libopencm3/stm32/f1/rtc.h or libopencm3/stm32/common/rtc_common_l1f024.h maybe?

karlp, 2019-09-25:
I can only ever find a rtc_common_bcd that was renamed to rtc_common_l1f024.  I've no idea what this might have been from.  The f3 rtc.h file was _added_ in 3c6e9fd56d in what appears to be unrelated.

That said.... f3 has the "bcd/v2" style rtc peripheral, (everyone except f1 has it) so... probably would "work" to replace it?

h2obrain, 2019-09-25:
Sorry, i do not have a f3 board to test this.

OpenUAS, 2019-09-27:
@h2obrain I can send you some F3 dev board for free, have some I don't use, just send and email where to

h2obrain, 2019-09-27:
@OpenUAS Thanks a lot for the offer, but I do not intend to start working with the f3-series. And having a board just to fix a bug I stumbled upon seems like a waste of a dev board. It should be quite easy to verify that rtc is working. I can have a look at the datasheet and provide some test-code.

karlp, 2020-01-04:
I've pushed 18f4d7c1 to fix this.  I don't have any f3 rtc code around, but it was obviously wrong, so I've just used the "correct" shared include.  Definitely better than before.  Thanks for reporting this!

closed 2020-01-04 by karlp