libopencm3/libopencm3 #1110: Memory corruption from build_config_descriptor

vielster, 2019-10-11:
Running on an STM32F427 with the `otgfs_usb_driver` the `build_config_descriptor()` function in usb_standard.c overruns the control buffer and corrupts memory of adjacent variables.

Some code reference below...

```c++
class Stm32UsbUart {
 public:
  /// These strings will show in the USB endpoint, but will also dictate how the ports show up under
  /// linux (/dev/serial/by-id/usb-<vendor>_<product>_<serial_number>-if00).
  struct UsbStrings {
    const char *vendor;         ///< Vendor string to report with endpoint.
    const char *product;        ///< Product string to report with endpoint.
    const char *serial_number;  ///< Serial Number for product to report.
  };
  /// Construct an STM32 usb uart. There should only be one of these in a system.
  /// @param strings  Strings to set into the endpoint for this device.
  Stm32UsbUart(const std::string &uart_name, const UsbStrings &strings) noexcept;
  ~Stm32UsbUart() noexcept = default;

  // Standard UART calls implemented by this driver.
  int Open();

  // Custom entrypoints for the implementation.
  const usbd_device *GetUsbDev() { return usbd_dev_; }
  void HandleIrq();  // Poll the usb device.

 private:
  // Callbacks that are wrapped directly from the C library, thus must be static.
  static void SetConfigCb(usbd_device *usbddev, uint16_t wValue);
  static void DataRxCb(usbd_device *usbd_dev, uint8_t ep);
  static void DataTxCb(usbd_device *usbd_dev, uint8_t ep);
  static usbd_request_return_codes ControlRequestCb(
      usbd_device *usbd_dev, struct usb_setup_data *req, uint8_t **buf, uint16_t *len,
      void (**complete)(usbd_device *usbd_dev, struct usb_setup_data *req));

  void HandleRx();   // Receive event callback.
  void HandleTx();   // Transmit event callback.
  void SetConfig();  // Config event callback.

  // Member variables.
  usbd_device *usbd_dev_ = nullptr;  // DUSB descriptor to use for USB requests.
  const char *usb_strings_[3];       // Local copy of the endpoint strings.

  // Software ringbuffers for TX and RX on the device.
  bool is_open_             = false;             // Are we already open.
  bool configured_          = false;             // Have we received configure callbackk yet.
  std::atomic_flag tx_busy_ = ATOMIC_FLAG_INIT;  // Track status of transmitter.
  uint8_t usbd_control_buffer_[64];

  // Make sure we don't copy this around since it should be a singleton.
  DISALLOW_COPY_AND_ASSIGN(Stm32UsbUart);
};

int Stm32UsbUart::Open() {
  if (is_open_) {
    // Do nothing.
  } else {
    /* Setup the USB Descriptor */
    usbd_dev_ = usbd_init(&otgfs_usb_driver, &cdcacm_dev_descriptor, &config, usb_strings_, 3,
                          usbd_control_buffer_, sizeof(usbd_control_buffer_));
    usbd_register_set_config_callback(usbd_dev_, SetConfigCb);
    is_open_    = true;

    // Setup an interrupt handler for USB
    nvic_set_priority(NVIC_OTG_FS_IRQ, 0xFF);  // Might want to increase priority?
    nvic_enable_irq(NVIC_OTG_FS_IRQ);
  }
  return 0;
}
```

Note that the `usbd_control_buffer_` is the last data member of the class. After creating this object, and calling `Open()`, the first call to `usbd_poll()` was causing my application to crash. In debugging, I set a watch point on the next variable declared _after_ this object, and it was getting hit by a `memcpy()` call in the `build_config_descriptor()` function in usb_standard.c. In stepping through, I found that 67 bytes were being written, so increasing the size of the control buffer to as little as 67 bytes alleviated this issue, but if the descriptor lengths, or number of descriptors, etc. ever change, I could see this issue coming back.

karlp, 2019-10-12:
the control buffer is user provided, and needs to be 2x descriptor size iirc.  64 is definitely way to small. where did you get that number from?  I can't remember the exact calculation on what packets outstanding are requried to fit in ther eplus the descriptor.

closed 2019-10-12 by karlp

vielster, 2019-11-07:
Thanks for the info on appropriate buffer size. My only remaining comment is that it would have been helpful if the stack had asserted or returned an error rather than corrupting memory.

karlp, 2019-11-07:
if you can suggest which apis you'd like to behave differenttly for this sort of thing that's useful  Assserts are little harder towork out how do them nicely within  a library context.