libopencm3/libopencm3 #1115: Issues with HSI init when moving from deprecated rcc_clock_setup_hsi

vielster, 2019-10-23:
Working with an stm32f302, attempted to move from deprecated rcc_clock_setup_hsi() to rcc_clock_setup_pll(), but clock doesn't start.

Looking at the code, it seems that these two lines are missing from the following `else{}` case.

```c++
rcc_set_sysclk_source(RCC_CFGR_SW_HSI); /* XXX: se cayo */
rcc_wait_for_sysclk_status(RCC_HSI);
```
https://github.com/libopencm3/libopencm3/blob/7be50a5e75ed2d163d38a6759347c5e778ac02ab/lib/stm32/f3/rcc.c#L370

There may be a more suitable way the maintainers would like this fixed, but this was my quick-fix.

karlp, 2019-10-24:
Could you please elaborate on what's going wrong for you here?  At reset, when these routines are used, f3s are already running on HSI, so there's no need to explicitly switch to them.  Even if you were running on another clock, you can just stay running on that clock until you switch to the PLL as the sysclock.  Can you share the clock struct you're using?

vielster, 2019-10-24:
So, while what you've stated above makes sense, it's not what I have observed on this part. I have not tested on any others, and most of our boards use the HSE oscillator, so this has not been an issue. I'm using the pre-defined clock configurations included in rcc.c to ensure the configuration isn't causing any issues, but with the modifications described, I've also gotten a 72MHz config working off of the HSI. 

When using this clock initialization code, the CPU boots and everything is happy:

```c++
  rcc_clock_setup_hsi(&rcc_hsi_configs[RCC_CLOCK_HSI_64MHZ]);
```

Since this call is marked as deprecated, I simply swapped the call over to:

```c++
  rcc_clock_setup_pll(&rcc_hsi_configs[RCC_CLOCK_HSI_64MHZ]);
```

The code doesn't boot, and when attaching to GDB it is stuck in the setup routine, specifically in an rcc_wait_for_osc_not_ready() call (here is some formatted GDB output for your viewing pleasure):

```
   │363     void rcc_clock_setup_pll(const struct rcc_clock_scale *clock)                                                                                                                                                                                    │
   │364     {                                                                                                                                                                                                                                                │
   │365             if (clock->pllsrc == RCC_CFGR_PLLSRC_HSE_PREDIV) {                                                                                                                                                                                       │
   │366                     rcc_osc_on(RCC_HSE);                                                                                                                                                                                                             │
   │367                     rcc_wait_for_osc_ready(RCC_HSE);                                                                                                                                                                                                 │
   │368             } else {                                                                                                                                                                                                                                 │
   │369                     rcc_osc_on(RCC_HSI);                                                                                                                                                                                                             │
   │370                     rcc_wait_for_osc_ready(RCC_HSI);                                                                                                                                                                                                 │
   │371             }                                                                                                                                                                                                                                        │
   │372             rcc_osc_off(RCC_PLL);                                                                                                                                                                                                                    │
   │373             rcc_usb_prescale_1_5();                                                                                                                                                                                                                  │
   │374             if (clock->usbdiv1) {                                                                                                                                                                                                                    │
   │375                     rcc_usb_prescale_1();                                                                                                                                                                                                            │
   │376             }                                                                                                                                                                                                                                        │
  >│377             rcc_wait_for_osc_not_ready(RCC_PLL);                                                                                                                                                                                                     │
   │378             rcc_set_pll_source(clock->pllsrc);                                                                                                                                                                                                       │
   │379             rcc_set_pll_multiplier(clock->pllmul);                                                                                                                                                                                                   │
   │380             rcc_set_prediv(clock->plldiv);                                                                                                                                                                                          [... cut ...]
I traced through the routine and saw that those two lines were in `rcc_clock_setup_hsi()` but not `rcc_clock_setup_pll()` and adding them alleviates the issue.

In order to leave my libopencm in a clean state, I've actually just changed my BoardInit() routine to:

```c++
  rcc_set_sysclk_source(RCC_CFGR_SW_HSI);  // workaround for issue with new PLL API.
  rcc_wait_for_sysclk_status(RCC_HSI);     // https://github.com/libopencm3/libopencm3/issues/1115
  rcc_clock_setup_pll(&rcc_hsi_configs[RCC_CLOCK_HSI_64MHZ]);
```

This also works fine.

karlp, 2019-10-24:
Are you sure you don't have anything else in your reset_handler or boardinit?

I've just tried this on both an old Rev Y stm32f303 on teh stm32f3discovery, and a newer part on a nucleo-f303re. (I don't have any f302 parts available on hand I'm afraid, nor can I find anything in theh docs that would indicate a relevant difference)

Is it possible your debugger is configuring clocks without your knowledge?  Some of them do to help "speed things up" eg: https://github.com/ntfreak/openocd/blob/master/tcl/target/stm32f3x.cfg#L87 
However, even with _that_ config, I still can't get this to fail.  Can you do something like my gdb session below to try and help work out what's going wrong here?

```
(gdb) mon reset halt
Unable to match requested speed 1000 kHz, using 950 kHz
Unable to match requested speed 1000 kHz, using 950 kHz
reset start
examine end....
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000980 msp: 0x20000800
(gdb) p /x RCC_CR
$3 = 0x5883    <<<< this is reset value, 0x58 are the part specific trim defaults
(gdb) call rcc_osc_off(RCC_PLL)  <<<< checking it doesn't do anything weird
Note: automatically using hardware breakpoints for read-only addresses.
(gdb) p /x RCC_CR
$4 = 0x5883
(gdb) call rcc_wait_for_osc_not_ready(RCC_PLL)   <<< checking it doesn't hang
(gdb) p /x RCC_CR
$5 = 0x5883
(gdb) break rcc_clock_setup_pll 
Breakpoint 1 at 0x8000648: file rcc.c, line 365.
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/karlp/src/libopencm3-miniblink/bin/stm32/stm32f3-discovery.elf 

Breakpoint 1, rcc_clock_setup_pll (clock=0x8000a50 <rcc_hsi_configs+24>) at rcc.c:365
365		if (clock->pllsrc == RCC_CFGR_PLLSRC_HSE_PREDIV) {
(gdb) p /x RCC_CR
$6 = 0x5883    <<<<< still clean at setup entry...
(gdb) p /x RCC_CFGR
$7 = 0x0
(gdb) c
Continuing.
halted: PC: 0x0800064a
^C    <<<< didn't hang, running blink code at this point.
```

For reference, and easy of testing, I did this on the https://github.com/libopencm3/libopencm3-miniblink repository, with the following patch:
```
diff --git a/template_stm32.c b/template_stm32.c
index 35ad03c..600a62f 100644
--- a/template_stm32.c
+++ b/template_stm32.c
@@ -3,6 +3,9 @@
 
 int main(void) {
         rcc_periph_clock_enable(RCC_LED1);
+#if defined(STM32F3)
+       rcc_clock_setup_pll(&rcc_hsi_configs[RCC_CLOCK_HSI_64MHZ]);
+#endif
 #if defined(STM32F1) /* F1 is a precious snowflake */
        gpio_set_mode(PORT_LED1, GPIO_MODE_OUTPUT_2_MHZ, GPIO_CNF_OUTPUT_PUSHPULL, PIN_LED1);
 #else /* everyone else is sane */

```

vielster, 2019-10-25:
So as a basic test, I flashed the device, noted the hung processor, disconnected the debugger and power cycled, and as you supposed, the system starts up just fine. It does appear that the debugger is touching the state and not cleaning it up. So it seems this workaround is only necessary while loading through the debugger. The old clock_setup_hsi was simply masking the issue. 

Will leave it to you if you think this is worth keeping in to prevent behavioral issues based on debugger/IDE/etc. (For reference, I'm using v1.5.1 of the stlink tools under linux (https://github.com/texane/stlink.git).

Thanks for the help and clarification. I will work around this as needed.

closed 2019-10-25 by vielster

karlp, 2019-10-25:
I recommend switching to openocd :)  openocd switches the PLL on so it can flash faster, but when you then run your own code, it resets it again so it's clean.  Your debugger should _not_ be touching stuff like that!