libopencm3/libopencm3 #1119: STM32F411CEU6 USB Device Not Working (otgfs_usb_driver maybe?)

fornellas, 2019-10-27:
I'm trying to get some stm32f4 exapmples working on this [STM32F411CEU6 board](https://www.aliexpress.com/item/4000103610226.html?spm=a2g0s.9042311.0.0.1fbc4c4dDiS2fQ), but no luck.

I'm using the example code for [/stm32/f4/stm32f4-discovery/](https://github.com/libopencm3/libopencm3-examples/tree/master/examples/stm32/f4/stm32f4-discovery/usb_cdcacm). Since STM32F411CEU6 can only go up to 100MHz and the crystal on the board is 25MHz, I just changed the clock setup to:

```
rcc_clock_setup_pll(&rcc_hse_25mhz_3v3[RCC_CLOCK_3V3_84MHZ]);
```

I inspected all the code, and all seem legit for STM32F411CEU6 (the IO ports seems correct), so I expected for it work. However, the Linux kernel won't log a single line when the board is connected.

Potential problems I believe I ruled out:

- **Hardware**: The board came with some demo firmware, with USB Serial builtin, that worked. Its ROM bootloader works fine for DFU upload. I also tested with other boards.
- **Clock**:  The datasheet (RM0383) tells: *The USB clock is used for driving the 48 MHz domain at fullspeed (12 Mbit/s) and must be enabled prior to configuring the OTG FS core.*. Given what I've seen in all other stm32/f4/ USB examples, this seems legit (despite going for 84MHz and not the maximum 100MHz).
- **Other USB Device Code**: I tested several other USB example codes from stm32/f4/, but same result.

I'm guessing the issue is with `otgfs_usb_driver`. It seems it is [aliased](https://github.com/libopencm3/libopencm3/blob/209dea96e7b45a1266b070bf1157a4e4a7a49800/include/libopencm3/usb/usbd.h#L59) for `stm32f107_usb_driver`, so maybe some stm32f1 vs stm32f4 difference there?

Thanks in advance for any light here.

karlp, 2019-10-27:
Probably duplicate of https://github.com/libopencm3/libopencm3/issues/1073 (please follow that and reply with the feedback still waiting there)

Ignore the f107 name, that's a red herring.  the "f107" just means "dwt_otg_fs"

fornellas, 2019-10-27:
I think this is something different than #1073 : there's no kernel message on enumeration failure whatsoever. dmesg is simply quiet.

Just in case, I also tested reverting 26ab78a, but same results.

Anything I can do to help debug this? Worst case, I can use my [logic analyzer](https://www.dreamsourcelab.com/product/dslogic-plus/) and share the captured data.

fornellas, 2019-10-27:
I tried the logic analyzer: D+ and D- are idle after powering the board on. Packet capture seems fine, as I can successfully capture and decode system memory DFU bootloader packets with the same setup. It seems whatever triggers USB communication to start, is not triggering.

fornellas, 2019-10-29:
@karlp any suggestions where to go from here? I'm suspecting the USB clock is not properly setup for my case. From what I've seen at the datasheet, clock setup is quite complex, I'd rather avoid digging through tens of pages of datasheet if there's something simpler you think I should check first. Thanks!

karlp, 2019-10-30:
You're currently in the best position to test it :)  I don't have any f411 (or f446) hardwar.  If you're not even getting enumeration though, your problems should be much simpler.  You _must_ check the clocks, to make sure that you are indeed providing the correct 48MHz clock to the usb periph, you _must_ check that any peripherals required for automatically doing the d+ pullup are enabled, if f411 needs anything special.  I've _very_ briefly skimmed the clock structure, and the pllq of 7 _seems_ correct, but you'll need to review this yourself.

fornellas, 2019-10-30:
Thanks a million, o7, will do!

fornellas, 2019-10-31:
> You must check the clocks, to make sure that you are indeed providing the correct 48MHz clock to the usb periph

I read through the datasheet and did the calculations: to the best of my knowledge this correctly sets up the USB OTG FS clock to 48MHz:
```
rcc_clock_setup_pll(&rcc_hse_25mhz_3v3[RCC_CLOCK_3V3_84MHZ]);
```
> you must check that any peripherals required for automatically doing the d+ pullup are enabled, if f411 needs anything special.

The Discovery boards used in the examples have a 1.5kOhm pull up on D+:

[image]

And supposedly work with this setup:

```
	rcc_periph_clock_enable(RCC_GPIOA);
	rcc_periph_clock_enable(RCC_OTGFS);

	gpio_mode_setup(GPIOA, GPIO_MODE_AF, GPIO_PUPD_NONE, GPIO11 | GPIO12);
	gpio_set_af(GPIOA, GPIO_AF10, GPIO11 | GPIO12);
```

The board I'm using apparently have no D+ pull up:

[image]

So I figured changing the setup to:

```
	rcc_periph_clock_enable(RCC_GPIOA);
	rcc_periph_clock_enable(RCC_OTGFS);

	gpio_mode_setup(GPIOA, GPIO_MODE_AF, GPIO_PUPD_NONE, GPIO11);
	gpio_mode_setup(GPIOA, GPIO_MODE_AF, GPIO_PUPD_PULLUP, GPIO12);
	gpio_set_af(GPIOA, GPIO_AF10, GPIO11 | GPIO12);
```

Would work, but no luck: idle D+/D- lines, just the same. I had a look ad the USB OTG FS part of the [datasheet](https://www.st.com/content/ccc/resource/technical/document/reference_manual/9b/53/39/1c/f7/01/4a/79/DM00119316.pdf/files/DM00119316.pdf/jcr:content/translations/en.DM00119316.pdf), it mentions about automatic hw pull up several times, not sure this software pull up would be needed.

@karlp thanks for the previous pointers, do you have any further suggestions here?

fornellas, 2019-10-31:
More investigation. The `otgfs_usb_driver` USB driver (which currently points to `stm32f107_usb_driver`) have [`stm32f107_usbd_init`](https://github.com/libopencm3/libopencm3/blob/564e485989fb8ba52c03a7f088cd59de41fe457e/lib/usb/usb_f107.c#L54) to handle its initialization sequence. It seems to do a different core initialization sequence that what the [datasheet tells](https://www.st.com/content/ccc/resource/technical/document/reference_manual/9b/53/39/1c/f7/01/4a/79/DM00119316.pdf/files/DM00119316.pdf/jcr:content/translations/en.DM00119316.pdf):

[image]
(there's also device initialization at device initialization at 22.17.3)

I'm suspecting that `stm32f107_usbd_init` implementation maybe the root cause for USB not to be working. I also checked other [stm32f4* datasheet](https://www.st.com/content/ccc/resource/technical/document/reference_manual/3d/6d/5a/66/b4/99/40/d4/DM00031020.pdf/files/DM00031020.pdf/jcr:content/translations/en.DM00031020.pdf), which also describe the same different initialization process.

@karlp  you mentioned that *"f107" just means "dwc_otg_fs"*, what's the story there? Are we missing an USB driver for STM32F411CEU6 or am I wrong here? If we're missing, I can cut a PR for that.

karlp, 2019-10-31:
f107 is the dwc_otg (this is the same usb core used on _many_ different platforms, it's a synopsis designware core)  it's used and tested on other f4s already and has been for some time.  Obviously there's _something_ that's not quite right, but these are all _meant_ to be code compatible.  By all means submit a fix if you can identify one, but it's by fixing the "f107" driver, not by making a new one.  There's tests in the tests/gadget-zero directory that you can use for regression testing.

fornellas, 2019-11-01:
I just confirmed this f107 driver is broken for STM32F411CEU6. I changed `stm32f107_usbd_init` to initialize things like this:

```c
	OTG_FS_GAHBCFG |= OTG_GAHBCFG_GINT;
	OTG_FS_GUSBCFG |= OTG_GUSBCFG_FDMOD | OTG_GUSBCFG_HNPCAP | OTG_GUSBCFG_SRPCAP | OTG_GUSBCFG_TRDT_MASK;
	OTG_FS_GINTMSK |= OTG_GINTSTS_OTGINT | OTG_GINTSTS_MMIS;
	OTG_FS_DCFG |= OTG_DCFG_DSPD;
	OTG_FS_GINTMSK |= OTG_GINTSTS_USBRST | OTG_GINTMSK_ENUMDNEM | OTG_GINTSTS_ESUSP | OTG_GINTSTS_USBSUSP | OTG_GINTSTS_SOF;
	OTG_FS_GCCFG |= OTG_GCCFG_NOVBUSSENS | OTG_GCCFG_PWRDWN;
```

and it works:

```
[46617.387684] usb 4-2: new full-speed USB device number 24 using ohci-pci
[46617.589974] usb 4-2: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
[46617.589981] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[46617.589985] usb 4-2: Product: CDC-ACM Demo
[46617.589989] usb 4-2: Manufacturer: Black Sphere Technologies
[46617.589992] usb 4-2: SerialNumber: DEMO
[46617.594458] cdc_acm 4-2:1.0: ttyACM0: USB ACM device
```

It is not quite there yet though. After `dfu-util` finishes programming and the device resets, the kernel can't talk to it:

```
[46585.370773] usb 4-2: new full-speed USB device number 23 using ohci-pci
[46600.787216] usb 4-2: device descriptor read/64, error -110
```

but it works after pressing the reset button or doing a power cycle.

After digging through all the register settings, I found some interesting things:

- OTG_GUSBCFG_FDMOD is forcing it in device mode. Without it, ideally, we should read the CMOD bit at OTG_FS_GINTSTS to confirm we're in device mode.
- OTG_FS_GUSBCFG / TOCAL: this calibration seems to be device dependent and may be required.
- OTG_FS_GUSBCFG / TRDT: It is dependent on the AHB frequency by following a table. I have it configured for 14.2-15MHz (OTG_GUSBCFG_TRDT_MASK), but it should be for 32MHz+, as my AHB frequency is 84MHz, but surprisingly it works.
- OTG_FS_GINTSTS: it should be cleared prior to unmasking interrupts.
- OTG_FS_DCFG / OTG_DCFG_NZLSOHSK: Datasheet says to configure, it seems that the default works... not sure about it.
- OTG_FS_GCCFG / OTG_GCCFG_VBUSBSEN: I had to use OTG_GCCFG_NOVBUSSENS instead for it to work with my board (likely due to the missing pull up).
- OTG_FS_GCCFG / OTG_GCCFG_PWRDWN: datasheet does not mention, but it is actually needed.

All that said, I have my doubts about fixing this by patching the "f107" driver:

1 - There are configurations that are board specific, like OTG_GCCFG_VBUSBSEN / OTG_GCCFG_NOVBUSSENS.
2 - Calibration is hardware specific, eg TOCAL.
3 - Given that the driver failed for my device, I wonder how many different uCs exist that it "should work" but does not. I expect the variety to be quite wide there, impossible to test / go through all datasheets. Perhaps having some pre-processor directives with a white list of devices know to work would help.

I think that 1 and 2 are enough to make `_usbd_driver` public, and allow people to customize stuff if needed, I'd be happy do to a PR for that.

Having the "f107" driver fixed is another issue. I'm learning as I go here, but it seems that there are way to many ifs needed in the code to make it work. I managed to get to 6 lines but `stm32f107_usbd_init` is already quite bigger than that. Not sure I can help much here.

fornellas, 2019-11-02:
Regarding the enumeration failure after `dfu-util`, [AN3156 at 5.5](https://www.st.com/content/ccc/resource/technical/document/application_note/6a/17/92/02/58/98/45/0c/CD00264379.pdf/files/CD00264379.pdf/jcr:content/translations/en.CD00264379.pdf) tells:

[image]

I added this to my usb init procedure:

```c
OTG_FS_GRSTCTL |= OTG_GRSTCTL_CSRST;
while (OTG_FS_GRSTCTL & OTG_GRSTCTL_CSRST);
while (!(OTG_FS_GRSTCTL & OTG_GRSTCTL_AHBIDL));
OTG_FS_GINTSTS = 0x04000020; // Reset value
```

But enumeration still fails. My guess is that the DFU firmware jumps to the application code leaving whatever configuration behind, that we should re-do, but it is gonna be a cat and mouse game to figure out what it is...

fornellas, 2019-11-02:
**Found the issue with `stm32f107_usbd_init`!**

```c
	if (OTG_FS_CID >= OTG_CID_HAS_VBDEN) {
		/* Enable VBUS detection in device mode and power up the PHY. */
		OTG_FS_GCCFG |= OTG_GCCFG_VBDEN | OTG_GCCFG_PWRDWN;
	} else {
		/* Enable VBUS sensing in device mode and power up the PHY. */
		OTG_FS_GCCFG |= OTG_GCCFG_VBUSBSEN | OTG_GCCFG_PWRDWN;
	}
```
`OTG_FS_CID` on STM32F411CEU6 is 0x00001200 and `OTG_CID_HAS_VBDEN` is 0x00002000, so this test fail and `OTG_GCCFG_VBUSBSEN` is set. If I change it to `OTG_GCCFG_VBDEN`, the `otgfs_usb_driver` works for STM32F411CEU6, despite the rest of the initialization not follow the datasheet, meh, but yeah!

The datasheet merely refers to `OTG_FS_CID` as "product id", what's the rational of comparing this value with `OTG_CID_HAS_VBDEN`? I tried doing `OTG_FS_CID = OTG_CID_HAS_VBDEN;` before `usbd_init()`, but it does not work (my guess is the core soft reset reverts the value).

From the datasheet:

[image]

So it seems VBUS sensing is dependent on how the board is built / powered, so we'd need a way to configure this. Making `_usbd_driver` public would allow that, but perhaps we can consider having `otgfs_novbussens_usb_driver` and be explicit to users instead.

@karlp how to you think we can make this work?

fornellas, 2019-11-09:
FYI, I managed to fix this in [my project](https://github.com/fornellas/3dpkbd2/blob/321ef6eafaa3616f3571fdfe88835f5d1c9dac16/keyboard/usb.c#L84-L95) by doing this:

```c
usbd_dev = usbd_init(
  &otgfs_usb_driver,
  ...       
);
OTG_FS_GCCFG |= OTG_GCCFG_NOVBUSSENS;
```
It would still be nice to have a library interface to configure NOVBUSSENS, or at least, have documentation regarding this.

rwilfong50401, 2019-12-01:
Thanks @fornellas for figuring this out. I too have started playing with the STM32F411CEU6 and your fix works for me. I did some tracing on the board and it looks like only DP and DM are connected from the USB connector. This makes sense because the board is made as close to the Bluepill design as possible. Also, somewhere above I believe there is mention of the pull-up on DP. Since the STM32F411 has a built in PHY that handles all the USB physical layer requirements, we don't need to worry about any external pull-up on DP.

Finally, based on my review of the specs, it looks like this board is limited to device mode, and would need different physical connections to the extra USB (PA8, PA9, PA10) lines for OTG host modes. That, and also a power injector.

fornellas, 2019-12-12:
@rwilfong50401 that seems to be the case.

The new issue I found, is that if the host issues an USB Reset (eg: when you power cycle the machine), enumeration won't work again: the device will stall all SETUP requests (I checked with a logic analyzer). I managed to go around this by soft resetting the whole device, but this is obviously far from optimal.

The code that handles USB Reset, just like the initialization code, does not follow what the datasheet states that must be done. Because it is the second time I find issues with this driver that piggy backs on stm32f107_usb_driver, I'm sketching a new otgfs_usb_driver, that I'll try to follow the datasheet as best as I can. USB has a tool https://www.usb.org/usb2tools to check for compliance, I think this would be the measure of success here :-D @rwilfong50401 , I'm glad to receive help if you can.

karlp, 2019-12-12:
I completely agree that the vbus sensing should be something that users should be able to select themselves.  as far as rewriting driver internals, yes, it's odd that it doesn't do the exact ordering that the  datasheet says.  If you want to re-order things, please make sure you're getting some testing on alternate hardware however, that's the purpose of the gadget0 tests, and teh reason we don't make changes here rapidly, "fixes" tend to be "breakages" at the same time.

Re the OTG_FS_CID check, that is checking for how to turn on vbus sensing for that version of the otg_fs core.  The different versions have the bit in a different place, with a different name.

fornellas, 2019-12-13:
o7 for gadget tests :-D

OTG_FS_CID: I'm confused to the "code compatibility for the family". Perhaps the differences you mentioned are for F1, and all F4 are such as the model on this post?

karlp, 2019-12-13:
there are different versions of f4, with different revisions in otg_fs_cid.  notably, the original f4 disco, and the f429 disco have different versions.  (you can git annotate to see where that all came from)

fornellas, 2019-12-14:
Oh, yeah, that's just the product ID, fair.

What I'm curios to know from your experience is, how does code
compatibility works across the same F4 family. There are over 140 models
with USB support
https://www.st.com/en/microcontrollers-microprocessors/stm32f4-series.html#products
And it would not be feasible to go through all of their datasheets.

Would it be safe to assume that if all tests passes for one model in the F4
family, it should work for others as well? I have only 2 different models
to test at hand.

fornellas, 2019-12-14:
@rwilfong50401 here's my WIP branch https://github.com/fornellas/libopencm3/blob/otgfs_usb_driver/lib/usb/usb_otgfs.c. I pulled code from dwg as a starting point, and am going through the datasheet and adding code accordingly. Should be straightforward, until you find reference for a register OTG_FS_TX0FSIZ that's nowhere to be found... meh.

rwilfong50401, 2019-12-14:
@fornellas As far as the different models of the F4 Family is concerned, they are similar, but they can be different. For example the 407 series has an OTG FS and OTG HS. The High Speed peripheral  doesn't have a built in PHY so it has special configuration. The Full Speed is very similar to the 411, except that the 407 has different versions of the PHY and registers. The documentation mentions the register changes in the revisions list at the end of the reference manual. Apparently if you had an older 407, you would need to use the old documentation to set the old registers correctly.

rwilfong50401, 2019-12-14:
@fornellas I looked at your changes so far, and it looks pretty good. I have a couple of comments.

First, if you are using the 411 reference manual RM0383 Rev 3, look at Figure 244 (USP peripheral-only connection.) Our board doesn't connect PA9, so it's correct that you use OTG_GCCFG_NOVBUSSENS. Since we also don't have connectivity to the ID line (PA10) I would also force peripheral-only mode (OTG_FS_GUSBCFG |= OTG_GUSBCFG_FDMOD). You might also consider disabling HNP and SRP as described in 22.3.2 and 22.4.3.

I only suggest this because it will make the OTG work more like the plain-old regular USB device. I get the feeling from looking through the libopencm3 code that this is the way things were originally set up, probably because OTG is relatively newer.

Just my 2 cents.

fornellas, 2019-12-14:
Thanks for the tips, I'll update my branch tomorrow with those.

ChuckM, 2019-12-15:
> Oh, yeah, that's just the product ID, fair. What I'm curios to know from your experience is, how does code compatibility works across the same F4 family. There are over 140 models with USB support [...]

Going through 140 data sheets is not "a lot of work" it is at most about a week of work. Just saying.  Generally though, you can just go through the various reference manuals, and there are only about 30 of those. The last time I did the 'every data sheet' thing was to figure out all possible pins that all possible SPI or I2C ports could appear on so that my SPI/I2C exercising code could configure any F4.

fornellas, 2019-12-15:
@rwilfong50401: I suppose since we're doing an USB device driver, going for OTG_GCCFG_NOVBUSSENS and OTG_GUSBCFG_FDMOD should work for all boards, right? HNP / SRP seems reasonable as well, but does not seem to be strictly needed. Regarding OTG FS/HS, I suppose we can (for the time being) focus on one of them (OTG FS), make it work, then see how much code can be shared with the other.

@ChuckM TBH, I'm happy if I make my F411 work well, and pass the USB tests. Going after the other 29 reference manuals is something that can be done after.

I updated my branch with more stuff. I put a lot of "TODO" that halts the code, and implementing them as they pop up.

I could use some help with 22.17.5, where it asks to program OTG_FS_TX0FSIZ. It is nowhere to be found, not sure WTH. I found this https://github.com/wookey-project/driver-stm32f4xx-usb/blob/master/stm32f4xx_usb_fs.c#L1216, which is my best guess at the moment.

rwilfong50401, 2019-12-15:
@fornellas I agree with al of your comments. In fact, I wouldn't worry about the OTG HS at all, because we have no way to test it unless we order a PHY component for HS.

I will check out the OTG_FS_TX0FSIZ thing. It sounds like it could be a buffer setting in the device descriptor for IN (Device to Host transmission). I'll get back to you on that, but based on your link I think you're on the right track.

rwilfong50401, 2019-12-15:
@fornellas I'm pretty sure OTG_FS_TX0FSIZ means OTG_FS_DIEPTXF0 according to Table 127. Note that in host mode the same register is called HNPTXFSIZ. Also this one is for device 0, and at the bottom of the table it has OTG_FS_DIEPTXFx for device IN endpoints 1 through 3. I think they changed the name along the way and didn't replace it in the instructions.

fornellas, 2019-12-21:
@rwilfong50401 do you have a clue about TX0FSA on OTG_FS_DIEPTXF0? Doc says "RAM start address", but it is a 16 bit value, and `void *` is 32 bit... at `usb_dwc_common.c` it is being set to `usbd_dev->driver->rx_fifo_size` (128).

fornellas, 2019-12-31:
> The new issue I found, is that if the host issues an USB Reset

@rwilfong50401 FYI, this issues is **not** related to libopencm3, I just fixed it with [this commit](https://github.com/fornellas/3dpkbd2/commit/88ef4038d97f57c27bd48a2aa09415207d9bacc6) on my project.

Regarding writing a new driver, while I managed to progress half way [on my branch](https://github.com/fornellas/libopencm3/tree/otgfs_usb_driver), I don't think there's need to pursue it any more. I just sent [this PR](https://github.com/libopencm3/libopencm3/pull/1155) to fix a bug in libopencm3, that was enough to make all USB compliance tests pass. So, despite the fact that the current driver does *not* follow the datasheet, it is fully working now, to the best of my knowledge.

@karlp it seems the only think missing to close this issue is deal with:

```
OTG_FS_GCCFG |= OTG_GCCFG_NOVBUSSENS;
```

being required for some boards in order to work. Given the driver is for device only, what do you think about adding this line to the current driver? I believe this is benign for all cases (unless some F4 chip does not support this...). @ChuckM  before digging through a lot of datasheets, do you think it will work?

fornellas, 2020-01-03:
PS: I just found out that on top of disabling `OTG_GCCFG_NOVBUSSENS`, I also had to:
`OTG_FS_GCCFG &= ~OTG_GCCFG_VBUSBSEN;`
Without it, PA9 (which is used for Vbus sensing and a/b device sensing) will half work: when you set its pullup, the voltage at the pin is ~1.1V, and is read with logic low. With `OTG_GCCFG_VBUSBSEN` disabled, pullup works, and we can read ~3.3V.

CC @rwilfong50401

karlp, 2020-01-04:
so. to clarify, I think we need an API method that allows controlling whether vbus sensing should be enabled or not.

Secondly, you need to be careful with those bits.  https://github.com/libopencm3/libopencm3/blob/master/include/libopencm3/usb/dwc/otg_common.h#L223-L226  bit 21 does different things on different core versions, and this then affects what A and B sensing do or do not.  This is exactly why there needs to be a proper api for enabling/disabling vbus sensing.

fornellas, 2020-01-09:
@karlp given the driver is device only, what things do you see we would loose by making OTG_GCCFG_NOVBUSSENS / OTG_GCCFG_VAUSBSEN  / OTG_GCCFG_VBUSBSEN disabled for all cases? Not sure I got this correctly, but I'm under the impression this would be benign for all cases (given that the uC specific bits are set).

v44r, 2020-05-23:
FWIW, the proposed OTG_FS_GCCFG |= OTG_GCCFG_NOVBUSSENS also fixes the similar "black pill 2" board based on 401CCU6 (had the same problem; no enumeration by linux).

yuanyanhui, 2020-09-21:
anyone got a working example of F412RET6? OTG_FS_GCCFG looks quite different to F411.