libopencm3/libopencm3 #1121: Anybody working on STM32H7 support?

davids5, 2019-10-28:
@karlp Forgive this being filed as an issue.  Do you know anybody working on STM32H7 support for libopencm3?

karlp, 2019-10-28:
@davids5 meet @westlicht, @westlicht, meet @davids5 
https://github.com/libopencm3/libopencm3/issues/1116 :)

closed 2019-10-28 by karlp

westlicht, 2019-10-28:
I actually decided to go with ST's HAL for the early hardware prototype I'm doing. I don't like their code much, but because this is also a new board design I don't want to bother too much with getting libopencm3 up and running. I first need working hardware and get there as quick as possible. The thing that put me off a bit is that STM32H7 has a somewhat more elaborate clock subsystem and I really didn't want to spend the time on this right now. Most of the F7 peripherals code should probably work out of the box. But I've not dig in too deep yet, so don't take my word :)

vielster, 2019-11-07:
I've started working on adding this support.  I have some basic pieces in place, including the top-level memory map, which should allow for a good deal of functionality (GPIO, SPI, DAC, etc.). Will be incrementally adding pieces to this.

I'd like to start pushing these upstream through appropriate PR/MR workflow. Should I associate with this ticket, or create new ones _and_ how do I actually create and push new dev branches into the repo?

karlp, 2019-11-07:
push to a branch on your own repo, and open a PR for it.  You can keep pushing changes to it and it will autoupdate.  We don't maintain any development branches within the main repo itself.