libopencm3/libopencm3 #1126: How to answer to USB Interrupt In Pipe Requests?

fornellas, 2019-11-09:
[USB HID Class](https://usb.org/sites/default/files/documents/hid1_11.pdf) spec requires for interrupt in pipes to be answered and I can't make it work.

The USB HID example seems to [register no callback for interrupts](https://github.com/libopencm3/libopencm3-examples/blob/master/examples/stm32/f1/lisa-m-1/usb_hid/usbhid.c#L259) and just let [systick send packets](https://github.com/libopencm3/libopencm3-examples/blob/master/examples/stm32/f1/lisa-m-1/usb_hid/usbhid.c#L398) without any regard for an interrupt having been received or not.

I tried setting up a callback:

```c
	usbd_ep_setup(
		dev,
		HID_ENDPOINT_IN_ADDR,
		USB_ENDPOINT_ATTR_INTERRUPT,
		HID_ENDPOINT_MAX_PACKET_SIZE,
		hid_endpoint_interrupt_in_callback
	);
```
I can see the interrupt in request via Wireshark:

[image]

but the callback is never invoked.

I had a look at `usb_ep_setup` / `dwc_ep_setup` / `dwc_poll` but could not spot any obvious issues there.

Did I miss something or this should have worked?

Thanks.

fornellas, 2019-11-09:
I experimented a bit, if I got this correctly, despite the doc not being clear, `usbd_ep_setup` sets up a callback for **after** the interrupt was served, not when it is generated as I had expected.

I managed to make the a keyboard work by:

- set a callback with `usbd_ep_setup`.
- `usbd_ep_write_packet`
- wait for the host to issue an interrupt request, which triggers the callback.
- Inside the callback, call `usbd_ep_write_packet` again.

I had a look a the datasheet for my uC, it seems there's a specific procedure to handle interrupt in requests, by waiting on TXFE on OTG_FS_DIEPINTx. I suppose it would be nice to have a callback for this case as well, so we can process the interrupt only when it comes, as opposed to preemptively populate the buffer, and wait an undetermined amount of time for the interrupt in to come from the host.

Makes sense?

fornellas, 2020-01-09:
I'm closing this issue. On second thought, USB is a host centric bus. When a device wants to send any interrupt, doing `usbd_ep_write_packet` and waiting for the host to fetch, is all we can do, so there's little value on implementing this extra interface.

closed 2020-01-09 by fornellas

karlp, 2020-01-10:
glad I could help :)