libopencm3/libopencm3 #1130: please limit the memcpy of usb endpoint descriptors to 7 bytes, to avoid garbage data

vampirefrog, 2019-11-16:
Hello, in usb_standard.c:117, there is a memcpy that would read garbage data (bytes from the *extra pointer) if you were to specify the endpoint length greater than 7.

Specifying a 9 byte endpoint length is sometimes needed when using a sync endpoint (the two extra bytes are bRefresh and bSynchAddress). 

It is currently impossible to specify a 9 byte endpoint, however with this quick fix it will be possible to stick the two extra bytes in the *extra data:

`memcpy(buf, ep, count = MIN(USB_DT_ENDPOINT_SIZE, MIN(len, ep->bLength)));`

You'll have to forgive me for not submitting a pull request.

vampirefrog, 2019-11-16:
And here would be some example code:

```C
static const struct {
	uint8_t bRefresh;
	uint8_t bSynchAddress;
	struct usb_audio_stream_audio_endpoint_descriptor audio_ep;
} __attribute__((packed)) audio_iso_playback_endp = {
	.bRefresh = 0,
	.bSynchAddress = 0x81,
	.audio_ep = {
		.bLength = sizeof(struct usb_audio_stream_audio_endpoint_descriptor),
		.bDescriptorType = USB_AUDIO_DT_CS_ENDPOINT,
		.bDescriptorSubtype = 0x01,
		.bmAttributes = 0x01,
		.bLockDelayUnits = 0,
		.wLockDelay = 0
	}
};

static const struct usb_endpoint_descriptor iso_playback_endpoints[] = {{
	.bLength = USB_DT_ENDPOINT_SIZE + 2,
	.bDescriptorType = USB_DT_ENDPOINT,
	.bEndpointAddress = 0x01,
	.bmAttributes = USB_ENDPOINT_ATTR_ISOCHRONOUS | USB_ENDPOINT_ATTR_ASYNC | USB_ENDPOINT_ATTR_DATA,
	.wMaxPacketSize = AUDIO_ENDPOINT_SIZE,
	.bInterval = 1,

	.extra = &audio_iso_playback_endp,
	.extralen = sizeof(audio_iso_playback_endp)
}, {
	.bLength = USB_DT_ENDPOINT_SIZE,
	.bDescriptorType = USB_DT_ENDPOINT,
	.bEndpointAddress = 0x81,
	.bmAttributes = USB_ENDPOINT_ATTR_ISOCHRONOUS | USB_ENDPOINT_ATTR_ASYNC | USB_ENDPOINT_ATTR_FEEDBACK,
	.wMaxPacketSize = 3,
	.bInterval = 10
}};
```

vampirefrog, 2019-11-16:
Or, perhaps, add the two extra bytes in the endpoint struct, because I assume `sizeof()` is never used on it:

```C
/* USB Standard Endpoint Descriptor - Table 9-13 */
struct usb_endpoint_descriptor {
	uint8_t bLength;
	uint8_t bDescriptorType;
	uint8_t bEndpointAddress;
	uint8_t bmAttributes;
	uint16_t wMaxPacketSize;
	uint8_t bInterval;
	uint8_t bRefresh;
	uint8_t bSynchAddress;

	/* Descriptor ends here.  The following are used internally: */
	const void *extra;
	int extralen;
} __attribute__((packed));
#define USB_DT_ENDPOINT_SIZE		7
#define USB_DT_ENDPOINT_WITH_FEEDBACK_SIZE		9
```

vampirefrog, 2019-11-16:
here is a couple of simpler test cases, without the audio nonsense:

This first case just intends to add two more bytes to the endpoint descriptor. However, the result you see in the screenshot. It is not reading the extra two bytes, but rather the first two bytes of the 32-bit address that's stored in the `extra` field of the struct, which is a pointer.

```C
#include <libopencm3/usb/usbd.h>
#include <libopencm3/cm3/scb.h>
#include <libopencm3/stm32/desig.h>
#include <libopencm3/stm32/rcc.h>
#include <libopencm3/stm32/gpio.h>
#include <libopencm3/usb/dwc/otg_fs.h>

//#include "usart.h"
#define debugf(fmt, ...)

#define MAX_PACKET_SIZE 192
#define ISO_OUT_ENDP_ADDR 0x01
#define ISO_IN_ENDP_ADDR 0x81

#define DEBUG_PORT GPIOC
#define DEBUG_PORT_RCC RCC_GPIOC
#define DEBUG_PIN  GPIO13

static const struct usb_device_descriptor dev = {
	.bLength = USB_DT_DEVICE_SIZE,
	.bDescriptorType = USB_DT_DEVICE,
	.bcdUSB = 0x0200,
	.bDeviceClass = 0,
	.bDeviceSubClass = 0,
	.bDeviceProtocol = 0,
	.bMaxPacketSize0 = 64,
	.idVendor = 0x6666,
	.idProduct = 0x9999,
	.bcdDevice = 0x0100,
	.iManufacturer = 1,
	.iProduct = 2,
	.iSerialNumber = 0,
	.bNumConfigurations = 1,
};

static const uint8_t extra_iso_bytes[] = { 0x55, 0x66 };

static const struct usb_endpoint_descriptor iso_out_endp[] = {{
	.bLength = 9,
	.bDescriptorType = USB_DT_ENDPOINT,
	.bEndpointAddress = ISO_OUT_ENDP_ADDR,
	.bmAttributes = USB_ENDPOINT_ATTR_ISOCHRONOUS,
	.wMaxPacketSize = MAX_PACKET_SIZE,
	.bInterval = 1,

	.extra = extra_iso_bytes,
	.extralen = 2
}};

static const struct usb_endpoint_descriptor iso_in_endp[] = {{
	.bLength = USB_DT_ENDPOINT_SIZE,
	.bDescriptorType = USB_DT_ENDPOINT,
	.bEndpointAddress = ISO_IN_ENDP_ADDR,
	.bmAttributes = USB_ENDPOINT_ATTR_ISOCHRONOUS,
	.wMaxPacketSize = MAX_PACKET_SIZE,
	.bInterval = 1,

	.extra = 0,
	.extralen = 0
}};

static const struct usb_interface_descriptor iso_out_ifaces[] = {{
	.bLength = USB_DT_INTERFACE_SIZE,
	.bDescriptorType = USB_DT_INTERFACE,
	.bInterfaceNumber = 0,
[... 237 lines cut ...]
```

[image]

In order to make this even clearer, let's use this simpler example:

```C
static const struct usb_endpoint_descriptor iso_out_endp[] = {{
	.bLength = 9,
	.bDescriptorType = USB_DT_ENDPOINT,
	.bEndpointAddress = ISO_OUT_ENDP_ADDR,
	.bmAttributes = USB_ENDPOINT_ATTR_ISOCHRONOUS,
	.wMaxPacketSize = MAX_PACKET_SIZE,
	.bInterval = 1,

	.extra = (void *)0xeeddccbb,
	.extralen = 0
}};
```

You will notice the bytes 0xbb 0xcc in the result:

[image]

I hope this clears things up. it is reading from the bytes of the pointer.

devanlai, 2019-11-16:
I have no particular stake in this issue, but here's a summary of the background information I've collected while looking into this for curiosity's sake that might be useful to anyone else whose confused about the state of endpoint descriptors.

- USB Audio and USB MIDI both define custom "data endpoint descriptors".
- However, there are actually two different types of endpoint descriptors:
  1. Standard XXX Data Endpoint Descriptors
  2. Class-specific XXX Data Endpoint Descriptors
- Standard Audio/MIDI data endpoint descriptors extend the standard USB descriptors. For Audio and MIDI, they both do so in the same way, by adding a `bRefresh` and `bSynchAddress` field to the end. So the resulting endpoint descriptor looks like a normal USB endpoint descriptor, but with a different `bLength` and more data at the end.
- Class-specific Audio/MIDI data endpoint descriptors are self-contained descriptors that follow the standard endpoint descriptor, but are otherwise self-contained with their own `bLength` and `bDescriptorType` fields.
- Support for class-specific descriptors is already handled for interface descriptors successfully in many places using the `extra`/`extralen` fields, and support for doing so for endpoint descriptors was added in [d3aa57](https://github.com/libopencm3/libopencm3/commit/d3aa579e0ae09fbcb58224b1ef83285a0e475c3e).
- As per [@svofski's comment](https://github.com/libopencm3/libopencm3/pull/603#commitcomment-15782195) on #603, there was dissention as to whether the existing `extra`/`extralen` mechanism works for "standard" endpoint descriptors.
- In the [existing MIDI sample code](https://github.com/ChuckM/stm32f429-discovery-examples/blob/e277f5004a093ab4d7d9a62997ced6b513f41200/apps/usb_midi/usbmidi.c#L89) in the examples repo, it uses the class-specific data endpoint descriptors, but not the "standard" data endpoint descriptors, just the actual unmodified data endpoint descriptors.

So my interpretation is that, as per this issue, locm3 doesn't really support the modified "standard" endpoint descriptors, and changes would need to be made to support them. I have no particular opinion as to whether @vampirefrog's proposed modifications to the mechanism for appending whole class-specific endpoint descriptors to handle appending extended data to standard endpoint descriptors is the best way to do this.

References:

- [USB Audio spec](https://www.usb.org/sites/default/files/audio10.pdf), Section 4.6.1: AS Isochronous Audio Data Endpoint Descriptors
- [USB MIDI spec](https://www.usb.org/sites/default/files/midi10.pdf), Sections 6.2.1 and 6.2.2: Standard MS Bulk Data Endpoint Descriptor, Class-Specific MS Bulk Data Endpoint Descriptor

vampirefrog, 2019-11-16:
Thanks for the refs

vampirefrog, 2019-11-16:
https://github.com/libopencm3/libopencm3/pull/1131

svofski, 2020-01-13:
Long time no see :) I'm trying to sync my project with the current version of libopencm3 right now. I see that audio.h now has the descriptors, this is nice. However I still don't see a reasonable solution for the 9-byte endpoint descriptors. My current rather ugly solution is to patch libopencm3 usbstd.h so that the standard usb_endpoint_descriptor contains `bInterval` and `bRefresh` and its size is 9.

@vampirefrog 's patch seems to offer a decent solution for my problem without being disruptive to the rest of the library.

vampirefrog, 2020-01-14:
The correct fix to this is to remove the *extra and extralen fields completely, which I've done here https://github.com/libopencm3/libopencm3/pull/1140

#1131 is a bit of a hack, but with #1140 you'll have to rewrite your descriptors, so pick your poison.

kgoba, 2021-03-17:
I am facing the same issue now with an audio class USB device. I cannot even build a proper USB configuration with the current implementation, as pointed out (due to inflexible bLength=7, where I need 9). I really like solution #1140; would be happy to see it implemented.