libopencm3/libopencm3 #1133: Allow USB descriptors to be stored fully in ROM

vampirefrog, 2019-11-17:
Hello. Currently it is not possible to define USB descriptors stored fully in ROM.

They are "compiled" into a RAM buffer, the control buffer, the 6th argument to `usbd_init()`. Currently it is necessary to have this buffer at least the size of the whole descriptor set. The compilation happens in `build_config_descriptor()`.

This compilation is currently needed because "`extra`" descriptors for interfaces and endpoints are pointers inside `struct usb_interface_descriptor` and `struct usb_endpoint_descriptor`.

There is also a `struct usb_endpoint_descriptor` `endpoint` pointer in `struct usb_interface_descriptor` that can be removed by simply concatenating the endpoint descriptor to the interface descriptor.

However, if the `extra` pointers (and `extralen` fields too) were to be removed, and if users would just concatenate the desriptor structs directly in ROM, then there would be no need for the `extra`, `extralen` fields and a large control buffer.

Here is a small example of how the user code would be transformed:

```C
static const struct usb_audio_stream_audio_endpoint_descriptor audio_iso_recording_endp[] = {{
	.bLength = 7,
	.bDescriptorType = USB_AUDIO_DT_CS_ENDPOINT,
	.bDescriptorSubtype = 0x01,
	.bmAttributes = 0,
	.bLockDelayUnits = 0,
	.wLockDelay = 0
}};

static const struct usb_endpoint_descriptor iso_recording_endp[] = {{
	.bLength = USB_DT_ENDPOINT_SIZE,
	.bDescriptorType = USB_DT_ENDPOINT,
	.bEndpointAddress = 0x82,
	.bmAttributes = USB_ENDPOINT_ATTR_ISOCHRONOUS | USB_ENDPOINT_ATTR_ADAPTIVE,
	.wMaxPacketSize = AUDIO_ENDPOINT_SIZE,
	.bInterval = 1,

	.extra = &audio_iso_recording_endp[0],
	.extralen = sizeof(audio_iso_recording_endp)
}};

```

The code above would become:

```C
static const struct {
	struct usb_endpoint_descriptor ep;
	struct usb_audio_stream_audio_endpoint_descriptor audio_ep;
} __attribute__((packed)) iso_recording_endp = {
	.ep = {
		.bLength = USB_DT_ENDPOINT_SIZE,
		.bDescriptorType = USB_DT_ENDPOINT,
		.bEndpointAddress = 0x82,
		.bmAttributes = USB_ENDPOINT_ATTR_ISOCHRONOUS | USB_ENDPOINT_ATTR_ADAPTIVE,
		.wMaxPacketSize = AUDIO_ENDPOINT_SIZE,
		.bInterval = 1
	},
	audio_ep: {
		.bLength = 7,
		.bDescriptorType = USB_AUDIO_DT_CS_ENDPOINT,
		.bDescriptorSubtype = 0x01,
		.bmAttributes = 0,
		.bLockDelayUnits = 0,
		.wLockDelay = 0
	}
};
```

However, this is a small example, but the entire config descriptor can be stored this way, fully in ROM.

This is how unicore-mx does it as well.

There is another advantage is that this way you can also specify the whole config descriptor as a byte array, which is useful for importing examples from other libraries and/or micros, [here](https://github.com/kiyoto-y/usb_dac_stm32f4/blob/master/Middlewares/ST/STM32_USB_Device_Library/Class/AUDIO/Src/usbd_audio.c#L177) is an example.

The disadvantage is that, besides the core library, all the examples in libopencm3-examples need to be rewritten, and probably tests and stuff. There is also another disadvantage, and that is that people could mistakenly order the descriptors wrong, whereas with the current method, there is a bit of restriction, but I'd argue that people can figure it out once they see that lsusb isn't listing their descriptors properly, and also people are smart enough to figure it out themselves.

I believe libopencm3 is the only one using this `extra` pointer method, and it is more a hindrance than an advantage.

Maybe @kuldeepdhaka can help too, as it would be quite the patch.

vampirefrog, 2019-11-17:
the .bLength can also be defined as `sizeof(struct usb_endpoint_descriptor)`, if we remove the `*extra` stuff

kuldeepdhaka, 2019-11-17:
Related:
https://github.com/libopencm3/libopencm3/issues/70
https://github.com/libopencm3/libopencm3/issues/1110

kuldeepdhaka, 2019-11-17:
There would be no need to have a buffer if string's are stored in rom.
that would save ram as no scratch buffer is required (and it can be big saving if someone is building something more than a led toggle)

Resulted code would be less, simpler, and better.

https://github.com/libopencm3/libopencm3/blob/master/lib/usb/usb_standard.c#L169
https://github.com/libopencm3/libopencm3/blob/master/lib/usb/usb_standard.c#L68-L140

For string:
https://github.com/libopencm3/libopencm3/blob/master/lib/usb/usb_standard.c#L171-L221

karlp, 2019-11-17:
no need for an issue for this, it's coverd by #70 already.

closed 2019-11-17 by karlp