libopencm3/libopencm3 #1144: LPC13xx Header Offsets Incorrect

colinoflynn, 2019-12-05:
It appears gpio.h for LPC13xx has incorrect offsets (https://github.com/libopencm3/libopencm3/blob/master/include/libopencm3/lpc13xx/gpio.h). They appear to need a further 0x8000 offset, see https://www.nxp.com/docs/en/user-guide/UM10375.pdf. Relevant table copied below:

[image]

As there appears to be a demo file wanted to check if there was some other offset coming in? Example code compiled here doesn't seem to work.

colinoflynn, 2019-12-05:
Edit - appears like https://github.com/ingobecker/libopencm3/blob/lpc1343/include/libopencm3/lpc13xx/gpio.h has this fix.

karlp, 2019-12-05:
We don't have any lpc13xx users, so I guess we trusted the submitter at the time, I would agree that the offsets look wrong, feel very welcome to submit a PR fixing them.

colinoflynn, 2019-12-05:
Thanks for confirming! Will submit PR for it then (wanted to check it wasn't some odd version or something first).