libopencm3/libopencm3 #1153: 【.gitignore】Please check this code: "include/libopencm3/**/nvic.h"

webees, 2019-12-18:
Obviously this "nvic.h" file should not be ignored.
[image]

webees, 2019-12-18:
https://github.com/libopencm3/libopencm3/issues/1096

karlp, 2019-12-19:
I'm afraid I still dont understand the problem you're trying to get across.

karlp, 2019-12-19:
If you're really trying to download a tarball, recreate it as a clean git repository from that point, you can just no include the .gitignore file yourself.  I'm not sure I follow what shoujld be changed in the project though.

webees, 2019-12-20:
I mean, this file: `https://github.com/libopencm3/libopencm3/blob/master/include/libopencm3/cm3/nvic.h` 
shouldn't be ignored, isn't it?

But the description of `.gitignore` is to ignore it. 
`https://github.com/libopencm3/libopencm3/blob/master/.gitignore`

```
# These are generated
include/libopencm3/**/nvic.h
include/libopencm3/**/**/nvic.h
```

Do you need to include the following files, or can these files be generated by recompilation.
```
!include/libopencm3/cm3/nvic.h
!include/libopencm3/dispatch/nvic.h
!include/libopencm3/lm4f/nvic.h
```

karlp, 2019-12-22:
Ok, I understand this issue now, thank you for clarifying

karlp, 2020-01-04:
thanks for pushing on this, I've just pushed 911d4be3 to cover this.  the other files you mark (and that git clean -xdn shows if you download the tarball and create a new git repo like you pointed out) are also important to be preserved, so I've included them all, with comments.

closed 2020-01-04 by karlp