libopencm3/libopencm3 #1154: how to handle -D_ROM=32K from genlink.py?

Jochen0x90h, 2019-12-18:
Hi!

executing the command
./genlink.py ../ld/devices.data STM32F042C6 DEFS

yields
-DSTM32F0 -DSTM32F042C6 -D_ROM=32K -D_RAM=6K ...

How can I handle "32K" in my c/cpp code? Shouldn't the genlink script replace K by multiplication with 1024, e.g. -D_ROM=32768?

Thanks!

Jochen0x90h, 2019-12-19:
What I want to achieve in the end is having a flash region for runtime programming of configuration data. Maybe this can also be done with variables that are explicitly placed into the .text or other section.
The luxury variant would be finding out which part of the flash is still empty, maybe by placing some dummy section at the very end of the flash area in the linker script.

karlp, 2019-12-19:
The K suffixes are understood by the linker when they are used in the linker scripts.  If you want to play with different sized sections for your own uses, then you are well and truly outside the realm of the linker script generator.  Something like taking the cortex-m-generic.ld as a base, using your own memory region definitions and adding whatever extra sections you like is your best option.

karlp, 2019-12-19:
I'm inclined to close this as "answered" but let me know if there's anything else that could be clarified here.

Jochen0x90h, 2019-12-20:
Then I have two feature requests (I know that pull requests are welcome ;) )
1. let genlink.py replace the K by *1024 for DEFS, because passing -D_ROM=32K instead of -D_ROM=32768 to the compiler seems to be not useful
2. by default add symbols to the linker script that lets you find the unused flash region at runtime, just like stack (and maybe heap) are placed in the unused ram region

karlp, 2019-12-25:
those are fine, but I would like stand by this as being "done" then.   For your hypothetical PRs, sure, 1 would be.... transparent to the linker, and if you can make that useful, sure, I guess.  Seems very thin, but *shrugs*  For 2, sure, but that's a PR for someone, not an issue we need to track.

closed 2019-12-25 by karlp

karlp, 2020-01-10:
do the comments at the end of https://github.com/libopencm3/libopencm3/issues/1117#issuecomment-570827827 help you at all?