libopencm3/libopencm3 #1159: lm4f and lm3s should have their own irq.json files

karlp, 2020-01-04:
Currently, the lm3s irq.json file defines a common superset of the irqs for the families, and then the lm4f nvic.h file includes the generated lm3s file.  This is.... functional, but there's no reason for the special casing, just having their own irq.json files would give them both perfectly suited files, with consistent generation, and no extra irq definitions.  Noticed originally while working on https://github.com/libopencm3/libopencm3/issues/1153

karlp, 2022-02-27:
well, it is an easy fix, if someone who's actually using these targets wants to fix.... :)

stefanct, 2022-02-27:
Probably, but I know my track record so I consciously did not promise anything in my previous reply ;) I have a patch to enhance the device database already though, so it is not out of the question!