libopencm3/libopencm3 #1164: Official way to use clang (llvm)?

denizzzka, 2020-01-17:
Please provide official way to use clang (llvm) for compiling whole libopencm3 project.

(Related issue: https://github.com/libopencm3/libopencm3/issues/244)

karlp, 2020-01-19:
So, what more do you need?  I'm not going to keep an issue around for it, but if you submit concrete changes taht make it work, that's more than welcome.  We're not at all opposed to clang, but someone has to do whatever work is needed.

closed 2020-01-19 by karlp

denizzzka, 2020-01-19:
Perhaps I did not properly understand the current state of clang support - I thought that everything is ready and it is just need to improve documentation.

karlp, 2020-01-20:
Maybe it is.  If there's something lacking, please provide what is needed.  What I mean is I'm not keeping a non-specific issue ticket like this around.  Does it not work? Open a ticket with something broken, Do you have the fix or improved docs? Open a PR, but I can't really work with "please provide this thing"

denizzzka, 2020-01-20:
> but I can't really work with

But maybe someone already did it, and then he can answer to this ticket or it will be a push to make PR

denizzzka, 2020-01-20:
Okay, forget it. Everyone needs to relax a bit!
Have a nice day!