libopencm3/libopencm3 #1172: stm32f1: add a struct based rcc clock setup like other families

karlp, 2020-02-03:
Leave the existing static functions, but mark them all deprecated.
Add a new struct based method similar to other stm32s.  Goal is for API unification, or at least consistency.  Fixes the last major issue in https://github.com/libopencm3/libopencm3/issues/266 and makes requests like https://github.com/libopencm3/libopencm3/issues/1171 easier

gollux, 2020-02-24:
Also, the current stm32f1 clock switching routines assume that they are run directly after reset. For example, if you call rcc_clock_setup_in_hsi_out_48mhz() and then rcc_clock_setup_in_hse_8mhz_out_72mhz(), clocks are misconfigured for two different reasons:

* The functions configure PLL without making sure that it is turned off first. (Documentation says that this must not be done.)
* Some clock setup functions configure the USB prescaler, but the others assume that the default state is correct.

It would be great if the clock unification fixed these problems. (It seems that the unified method handles that, at least when I looked at stm32f4.)

karlp, 2020-02-26:
it's a common problem yes, scaling up and down isn't always handled correctly, it was partly addressed for one family here: 095ed851 but it's definitely not robustly implemented for every possible use case.

closed 2020-12-01 by karlp