libopencm3/libopencm3 #1174: command not found while building software

kbickham, 2020-02-05:
I'm using the book "Beginning STM32" to learn the Embedded ARM Toolchain and starting using that from a Linux environment, in attempts to move past the arduino ide.

I have ubuntu running on a vm. I unpacked the cross compiler, renamed directly to gcc-arm, checked type for gcc and cross compiler... and it seems okay. Checking version, it seems ok.

The only thing I think that needs clarification is the alias armpath I added to the .bashrc file,
 which is 
alias armpath='export PATH="/opt/gcc-arm/bin:$PATH"'  ..which seems to work.

Initially running make, it gave me a problem with python not being installed...but said that python3 was. I wasn't sure if It'd be more work to go through all of these new tools and change arguments to python3 so... i ran:

sudo apt install python (for version 2.7)

When I run make now... I get this (image attached). I'm attempting to build the software..any idea what I'm doing wrong, or how i might going about fixing the situation?

[image]

grevaillot, 2020-02-05:
Hi,

It seems that you attempt to build as root, using sudo, to avoid some permissions issues : sudo does not takes care of your previously defined path. You need to fix your permissions issues properly, and run make without sudo :)

kbickham, 2020-02-05:
Wow...that's what I get for being lazy. Thank you.

closed 2020-02-05 by kbickham

karlp, 2020-02-05:
(related, our python scripts run with either python2 or python3, so shouldn't matter, but unfortunately, despite following the PEP for dual version scripts, not all distros are handling it properly)

kbickham, 2020-02-05:
I was using the latest ubuntu image from osboxes.org, just to verify. I did apt-get install the cross compiler from the terminal again, and after adding to admin and sudo groups (even though sudo seemed to be functioning before) it built correctly.