libopencm3/libopencm3 #1187: Pin remapping for STM32F103RCT6

krish2487, 2020-03-17:
Hello again, 

I am trying to use the PA15 pin on the STM32F103RCT6 as a GPIO.

I understand from the pinouts that the particular pin is assigned to JATG on power up and needs to be remapped to use it as a GPIO. 
From what I read, these are the sequence of steps I need to follow to remap the pin to use as a GPIO.

1. Enable AFIO clock.
2.Enable peripheral clock.
3. remap pins (turn off the JTAG / SWD function and enable the AFIO) .
4. set gpio output mode options.

And this is the entirety of the code to test if the remapping works and if I am able to toggle the pin high and low..

```
#include "libopencm3/stm32/rcc.h"
#include "libopencm3/stm32/gpio.h"
#include "libopencm3/stm32/pwr.h"

static inline __attribute__((always_inline)) void __WFI(void)
{
    __asm volatile ("wfi");
}

int main(void)
{
rcc_clock_setup_in_hsi_out_24mhz();

//Enable GPIOC Peripheral clock
rcc_periph_clock_enable(RCC_GPIOB);
rcc_periph_clock_enable(RCC_AFIO);
rcc_periph_clock_enable(RCC_GPIOA);

// Full remap or partial remap 01
// TIM2_REMAP[1:0]:TIM2remapping
// These bits are set and cleared by software. They control the mapping of TIM2 channels 1 to 4 and external trigger (ETR) on the GPIO ports.
// 00: No remap (CH1/ETR/PA0, CH2/PA1, CH3/PA2, CH4/PA3)
// 01: Partial remap (CH1/ETR/PA15, CH2/PB3, CH3/PA2, CH4/PA3)
// 10: Partial remap (CH1/ETR/PA0, CH2/PA1, CH3/PB10, CH4/PB11) 
// 11: Full remap (CH1/ETR/PA15, CH2/PB3, CH3/PB10, CH4/PB11)
gpio_primary_remap(AFIO_MAPR_SWJ_CFG_JTAG_OFF_SW_OFF, AFIO_MAPR_TIM2_REMAP_FULL_REMAP);

//Output mode, no pull ups or pull down
gpio_set_mode(GPIOB, GPIO_MODE_OUTPUT_10_MHZ, GPIO_CNF_OUTPUT_PUSHPULL,GPIO0 | GPIO1 | GPIO14);
gpio_set_mode(GPIOA, GPIO_MODE_OUTPUT_10_MHZ, GPIO_CNF_OUTPUT_ALTFN_PUSHPULL,GPIO15);

//Set GPIO8 and clear GPIO9 to see toggle
    gpio_set(GPIOB, GPIO1);
    gpio_clear(GPIOB, GPIO0);
    gpio_set(GPIOB, GPIO14);
    
    // Pin voltage must read about 3.3V.
    gpio_set(GPIOA, GPIO15);
    while (1)
        {
            volatile uint32_t i = 0;

            /* code */
            gpio_toggle(GPIOB, GPIO0 | GPIO1);

            //The counter i must be declared 'volatile' to prevent compiler optimization
            for (i = 0; i < 20000000; i++)
                {
                    __asm__("nop");
                }
        gpio_clear(GPIOB, GPIO14);

        //Clearing the Pin here. must go low.
        gpio_clear(GPIOA, GPIO15);

        for (i = 0; i < 2000000; i++)
        {
            __asm__("nop");
        }
        gpio_toggle(GPIOB, GPIO0 | GPIO1);
        pwr_set_stop_mode();
         __WFI();
        }

    return 0;
}

```

Unfortunately, the pin seems to be sitting at around 300mV and not changing at all.
It does not toggle.

I have tried "AFIO_MAPR_TIM2_REMAP_FULL_REMAP" and also "AFIO_MAPR_TIM2_REMAP_PARTIAL_REMAP1" to see if that makes a difference.

It did not.

Am i doing something obviously wrong ?? or did I mess up the initialization sequence or something else??

Any help is appreciated!! Thanks in advance!! :-)

karlp, 2020-03-17:
No part of your code is attempting to toggle A15?  The timer remap seems irrelevant, if you're only after A15 as a gpio, but I've not looked at the remap code closely.

krish2487, 2020-03-17:
Thank you karl.

```
 // Pin voltage must read about 3.3V.
    gpio_set(GPIOA, GPIO15);

elsewhere...
//Clearing the Pin here. must go low.
        gpio_clear(GPIOA, GPIO15);

```

It is not toggling the GPIO A15. I am attempting to turn it high. wait for a while and then switch low.

I expect to see 3.3V for about 8-10 seconds and then 0V on the pin. 

Regarding the timer remap, That is the only relevant remap I could see in the header files and in the pin outs. I have tried just using the Alternate Functions without remapping but it remains the same. 

I am at a loss as to what other setting I should configure it for?? 

I would be grateful for any suggestions..

Thank you again! :-)

Tony-LYX, 2020-03-17:
Well, I am also one of the user of libopcm3.
And also I am one of the user of "remap"
But unfortunately, I can't find a function called "remap", beacause the chip i use is F4 series.
Actually, after noticing your issue, I searched the lib and the document.
Only to find that "remap" fucntion only exist in "stm32/F1" folder.

I guess, the "remap" function is actually doesn't exist.
you are using a funtion which only have a "name".
if you really need "remap" function, maybe the official lib is a better choice

krish2487, 2020-03-17:
OK. got it working now..
@karlp  Your hint about not needing remapping in the first place was spot on!! :-)
Thanks..

Turns out, I needed to.
1.Enable the AFIO and peripheral clocks.
2.Turn off the JTAG in the MAPR
3.Configure the GPIO in the usual fashion.

Did I get the flow right??

So this is what works for me to control the pin PA15 as a GPIO.
And the relevant code snippet is 

```
rcc_clock_setup_in_hsi_out_24mhz();

//Enable GPIOC Peripheral clock
rcc_periph_clock_enable(RCC_GPIOB);
rcc_periph_clock_enable(RCC_AFIO);
rcc_periph_clock_enable(RCC_GPIOA);

AFIO_MAPR |= AFIO_MAPR_SWJ_CFG_JTAG_OFF_SW_ON;

//Output mode, no pull ups or pull down
gpio_set_mode(GPIOB, GPIO_MODE_OUTPUT_10_MHZ, GPIO_CNF_OUTPUT_PUSHPULL,GPIO0 | GPIO1 | GPIO14);
gpio_set_mode(GPIOA, GPIO_MODE_OUTPUT_10_MHZ, GPIO_CNF_OUTPUT_PUSHPULL,GPIO15);

//Set GPIO8 and clear GPIO9 to see toggle
    gpio_set(GPIOB, GPIO1);
    gpio_clear(GPIOB, GPIO0);
    gpio_set(GPIOB, GPIO14);
    gpio_set(GPIOA, GPIO15);
```

@karlp Thanks again!! I am closing this issue now.

closed 2020-03-17 by krish2487

karlp, 2020-03-17:
@Tony-LYX remap is an f1 specific thing.  F1 is the only stm32 with that gpio peripheral.

However, the function is actually pretty broken. (Don't try calling it twice to remap something a second time!) See https://github.com/libopencm3/libopencm3/issues/503

However for just "I want to get PA15 back as GPIO" you should be able to just do.

```gpio_primary_remap(AFIO_MAPR_SWJ_CFG_JTAG_OFF_SW_ON, 0)```
or
```gpio_primary_remap(AFIO_MAPR_SWJ_CFG_JTAG_OFF_SW_OFF, 0)``` (if you are so hardcore you don't ever want to debug again, and are sure you can connect under reset)

The second parameter is for the "other" remaps.  It's not a good function, but it doesn't do much, so it's fairly easy to figure [out from the docs](
http://libopencm3.org/docs/latest/stm32f1/html/group__gpio__file.html#ga793f728ea95a83f58e1d27d07d2fe0eb) (there's a link to the source there too)

As for your demo code, you're setting it to altfunction, but then trying to use it as a gpio.  I don't believe that will do what you want.  If you're trying to use it as a gpio, then you need to use it as a gpio, just like you do with port B.  if you really _do_ need to use it as a peripheral, then you need to do _that_ not play with it as a gpio. (And I think this is what you really want, as you seem to be trying to setup timer channels)

krish2487, 2020-03-17:
@karlp  Thank you for the handholding!! 👍 
I really owe a LOT to this awesome library and community and folks like you. 
It explains a lot!!

I seem to be very proficient in explaining my situation wrong.. :-D

You have it right with the first option. I really am trying to use the pin PA15 as a GPIO.
I, mistakenly chose the timer remap as it was given in the datasheet pinouts. That was my error of understanding. I _do_ want to use the GPIO functionality.

Without this line
`rcc_periph_clock_enable(RCC_AFIO);`

I am unable to turn the pin high and low. It sits, as I explained in the original post, at around 300 mV.
The 
`gpio_primary_remap(AFIO_MAPR_SWJ_CFG_JTAG_OFF_SW_ON, 0)`
works just fine. But to control the pin as a GPIO, I still am needing to enable the AFIO clock.

karlp, 2020-03-17:
Yes, to use the AFIO_MAPR register to get A15 back from jtag, you need to clock the AFIO peripheral :)  Glad you got it worked out,sorry about the delay, I'd actually started my reply here earlier in the day, and then got distracted and never saw your update before I finished posting:)

krish2487, 2020-03-17:
No worries.. as I said. Its an awesome library and you have been very helpful!!