libopencm3/libopencm3 #1192: isolating USB code

gimbas, 2020-03-25:
I have a project using the stm32f103 microcontroller in which i need a usb hid implementation, unfortunately there doesn't seem to be a lot of usb only libraries/implementations for this device out there, or at least readily available.

libopen-cm3 seems pretty capable for what i want, but it has way more than i need, i would like to isolate the usb implementation for the stm32f103, is it very dependent on the rest of the project? i would like some pointers.

karlp, 2020-03-26:
do you have an particular problems?  Have you tried just using it, and letting the compiler and linker sort out what you are and aren't using?  It's not (at all) a goal of this project to try and make standalone pieces I'm afraid, so unless you've got something specific, there's not really much else that can be said here.

closed 2020-03-26 by karlp

gimbas, 2020-03-26:
I have not tried that, no. But I didn't want to include code I am not using. just tell me what does the usb stack depend on? system tick? gpios hal?