libopencm3/libopencm3 #1193: Bad naming lib/stm32/f4/stm32f405x6.ld

UweBonnes, 2020-03-27:
Hello,
there is a linker file
lib/stm32/f4/stm32f405x6.ld
but it is for stm32f405xg and stm32f407xg. This needs to get renamed. 
However I am not sure if to use stm32f40zxg or stm32f40xzg. Please advice how to name or rename yourself.

Bye

karlp, 2020-03-27:
Agreed, the file is wrongly named.  However, renaming it is pointless, as it's the only one in the f4 family even included.
My general feeling on this is to simply remove them all.  we've not acceppted new ones for a long time now, and we've provided the linker script generator to handle the massive amount of parts available, and we have a generic section file that can be included, lib/cortex-m-generic.ld that only needs the user to provide their own region sizes file, eg
```
MEMORY
{
	rom (rx) : ORIGIN = 0x08000000, LENGTH = 256K
	ram (rwx) : ORIGIN = 0x20000000, LENGTH = 48K
}

/* Include the common ld script. */
INCLUDE cortex-m-generic.ld
```
So the usefulness of the (very limited selection) of prebaked files is highly doubtful.

UweBonnes, 2020-03-30:
Karl Palsson writes:
> Agreed, the file is wrongly named. However, renaming it is pointless, as it's
> the only one in the f4 family even included.
> My general feeling on this is to simply remove them all. we've not [...]
In Ethernut I solved this by providing giving the memory region size
on the linker command line like (for a F767): 
arm-none-eabi-gcc gpio.o -Wl,--defsym=itcm_length=16k \
-Wl,--defsym=flash0_length=2048K -Wl,--defsym=ccm_length=0 \
-Wl,--defsym=ram0_length=128k -Wl,--defsym=ram1_length=368K \
-Wl,--defsym=ram2_length=16K

and the linker file starts like:
ENTRY(NutInit)
SEARCH_DIR(.)
SRAM1_START = DEFINED(ram1_start) ? ram1_start : 0x20000000  + ram0_length;
SRAM2_START = SRAM1_START + ram1_length;

MEMORY
{
    ITCM  (rwx) : ORIGIN = 0x00000000, LENGTH = itcm_length
    ITCMFL (rx) : ORIGIN = 0x00200000, LENGTH = flash0_length
    FLASH0 (rx) : ORIGIN = 0x08000000, LENGTH = flash0_length
    SRAM0 (rwx) : ORIGIN = 0x20000000, LENGTH = ram0_length
    SRAM1 (rwx) : ORIGIN = SRAM1_START, LENGTH = ram1_length
    SRAM2 (rwx) : ORIGIN = SRAM2_START, LENGTH = ram2_length
}

Bye

-- 
Uwe Bonnes                bon@elektron.ikp.physik.tu-darmstadt.de

Institut fuer Kernphysik  Schlossgartenstrasse 9  64289 Darmstadt
--------- Tel. 06151 1623569 ------- Fax. 06151 1623305 ---------