libopencm3/libopencm3 #1194: How to use libopencm3 as a submodule with Meson build system?

vikaig, 2020-03-27:
Right now I'm just using this package https://www.archlinux.org/packages/community/any/libopencm3/ and the meson `find_library()` method(example https://github.com/vikaig/stm32-libopencm3-meson-blink/blob/master/meson.build#L25), but this package is only available in Arch Linux, it's not in other distributions and the manual installation of the library has been removed, so I want to switch to a submodule, but I don't understand what I should use instead of `find_library()` for a submodule? `subproject()` and `dependency()` will not work, libopencm3 does not have Meson or CMake support.

karlp, 2020-03-27:
if you want to contribute user documentation on how to do this, that would be welcome, just file a PR with it.  I have no experience with meson, and only generally poor experiences with cmake, so it's not something I'm spending time on personally.

closed 2020-03-27 by karlp

denizzzka, 2020-10-01:
@vikaig I sucessfully use libopencm3 from Meson

Feel free to improve this script to more common support of various MCUs: 
https://github.com/denizzzka/libopencm3/blob/meson_build/meson.build

denizzzka, 2020-10-01:
But it would be better to write native Meson support, of course.