libopencm3/libopencm3 #1195: How to use HID RAW

tadryanom, 2020-03-30:
Hello people!

I'm trying to implement the USB HIDRAW interface using libopencm3 with the stm32103c8t6 Blue Pill card, however, I'm not getting it. My data structures are these:

#include <libopencm3/usb/usbd.h>
#include <libopencm3/usb/usbstd.h>
#include <libopencm3/usb/hid.h>
#include <libopencm3/stm32/st_usbfs.h>

static const struct usb_device_descriptor dev = {
	.bLength = sizeof(struct usb_device_descriptor),
	.bDescriptorType = 1,
	.bcdUSB = 0x0200,
	.bDeviceClass = 0,
	.bDeviceSubClass = 0,
	.bDeviceProtocol = 0,
	.bMaxPacketSize0 = 64,
	.idVendor = 0x16c0,
	.idProduct = 0x0480,
	.bcdDevice = 0x0100,
	.iManufacturer = 1,
	.iProduct = 2,
	.iSerialNumber = 3,
	.bNumConfigurations = 1,
};

static const uint8_t rawhid_hid_report_desc[] = {
	0x06, 0xAB, 0xFF, // usage_page - recommended: 0xFF00 to 0xFFFF
	0x0A, 0x00, 0x02, // usage recommended: 0x0100 to 0xFFFF
	0xA1, 0x01,				// Collection 0x01
	0x75, 0x08,				// report size = 8 bits
	0x15, 0x00,				// logical minimum = 0
	0x26, 0xFF, 0x00,			// logical maximum = 255
	0x95, 0x40,			// report count
	0x09, 0x01,				// usage
	0x81, 0x02,				// Input (array)
	0x95, 0x40,			// report count
	0x09, 0x02,				// usage
	0x91, 0x02,				// Output (array)
	0xC0,					// end collection
};

static const struct {
    struct usb_hid_descriptor header;
    uint8_t bDescriptorType;
    uint16_t wDescriptorLength; 
    uint8_t bDescriptorTypeOpt;
} __attribute__((packed)) hid = {
    .header = {
        .bLength = 9,
        .bDescriptorType = 0x21,
        .bcdHID = 0x0111,
        .bCountryCode = 0,
        .bNumDescriptors = 1,
    }, 
    0x22,                           // bNumDescriptors
    sizeof(rawhid_hid_report_desc),	// wDescriptorLength
    0, 
};

static const struct usb_endpoint_descriptor data_endp[] = {
	{
		.bLength = 7,
		.bDescriptorType = 5,
		.bEndpointAddress = 0x81,		/* To Host */
		.bmAttributes = 0x03,
		.wMaxPacketSize = 64,
		.bInterval = 2,
	}, {
		.bLength = 7,
		.bDescriptorType = 5,
		.bEndpointAddress = 0x02,		/* From Host */
		.bmAttributes = 0x03,
		.wMaxPacketSize = 64,
		.bInterval = 8,
	}
};

static const struct usb_interface_descriptor data_iface[] = {
	{
		.bLength = 9,
		.bDescriptorType = 4,
		.bInterfaceNumber = 0,
		.bAlternateSetting = 0,
		.bNumEndpoints = 2,
		.bInterfaceClass = 0x03,
		.bInterfaceSubClass = 0,
		.bInterfaceProtocol = 0,
		.iInterface = 0,
        
        .endpoint = data_endp,
        
        .extra = &hid,
		.extralen = 9,
	}
};

static const struct usb_interface ifaces[] = {
	{
		.num_altsetting = 1,
		.altsetting = data_iface,
	}
};

static const struct usb_config_descriptor config = {
	.bLength = 9,
	.bDescriptorType = 2,
	.wTotalLength = 41,
	.bNumInterfaces = 1,
	.bConfigurationValue = 1,
	.iConfiguration = 0,
	.bmAttributes = 0xC0,
	.bMaxPower = 50,
	.interface = ifaces,
};

I used as reference those structures that work in the AVR architecture, respectively in an ATMEGA32U4, as follows:

static const uint8_t device_descriptor[] = {
	18,					// bLength
	1,					// bDescriptorType
	0x00, 0x02,				// bcdUSB
	0,					// bDeviceClass
	0,					// bDeviceSubClass
	0,					// bDeviceProtocol
	32,				// bMaxPacketSize0
	0xC0, 0x16,		// idVendor
	0x80, 0x04,	// idProduct
	0x00, 0x01,				// bcdDevice
	1,					// iManufacturer
	2,					// iProduct
	0,					// iSerialNumber
	1					// bNumConfigurations
};

static const uint8_t rawhid_hid_report_desc[] = {
	0x06, 0xAB, 0xFF,
	0x0A, 0x00, 0x02,
	0xA1, 0x01,				// Collection 0x01
	0x75, 0x08,				// report size = 8 bits
	0x15, 0x00,				// logical minimum = 0
	0x26, 0xFF, 0x00,			// logical maximum = 255
	0x95, 64,			// report count
	0x09, 0x01,				// usage
	0x81, 0x02,				// Input (array)
	0x95, 64,			// report count
	0x09, 0x02,				// usage
	0x91, 0x02,				// Output (array)
	0xC0					// end collection
};

static const uint8_t config1_descriptor[41] = {
	// configuration descriptor, USB spec 9.6.3, page 264-266, Table 9-10
	9, 					// bLength;
	2,					// bDescriptorType;
	0x29, 0x00,			// wTotalLength
	1,					// bNumInterfaces
	1,					// bConfigurationValue
	0,					// iConfiguration
	0xC0,					// bmAttributes
	50,					// bMaxPower

	// interface descriptor, USB spec 9.6.5, page 267-269, Table 9-12
	9,					// bLength
	4,					// bDescriptorType
	0,			// bInterfaceNumber
	0,					// bAlternateSetting
	2,					// bNumEndpoints
	0x03,					// bInterfaceClass (0x03 = HID)
	0x00,					// bInterfaceSubClass (0x01 = Boot)
	0x00,					// bInterfaceProtocol (0x01 = Keyboard)
	0,					// iInterface
	// HID interface descriptor, HID 1.11 spec, section 6.2.1
	9,					// bLength
	0x21,					// bDescriptorType
	0x11, 0x01,				// bcdHID
	0,					// bCountryCode
	1,					// bNumDescriptors
	0x22,					// bDescriptorType
	sizeof(rawhid_hid_report_desc),		// wDescriptorLength
	0,
	// endpoint descriptor, USB spec 9.6.6, page 269-271, Table 9-13
	7,					// bLength
	5,					// bDescriptorType
	0x81,		// bEndpointAddress
	0x03,					// bmAttributes (0x03=intr)
	64, 0,			// wMaxPacketSize
	2,			// bInterval
	// endpoint descriptor, USB spec 9.6.6, page 269-271, Table 9-13
	7,					// bLength
	5,					// bDescriptorType
	0x02,			// bEndpointAddress
	0x03,					// bmAttributes (0x03=intr)
	64, 0,			// wMaxPacketSize
	8			// bInterval
};

What could I be doing wrong?

Grateful for the attention.

mewhpm, 2020-08-02:
See https://github.com/konachan700/Mew/tree/master/bootloader/drivers/usb - it works well

karlp, 2020-08-06:
Closing as there's been a reply, please re-open if you don't have enough information now.

closed 2020-08-06 by karlp