libopencm3/libopencm3 #1200: Example for stm32f429i-discovery doesn't compile

kristofmulier, 2020-04-06:
I've cloned the Git repo:
https://github.com/libopencm3/libopencm3-examples.git

The cloned repo has an empty folder `libopencm3`. So I've cloned the following Git repo in that empty folder:
https://github.com/libopencm3/libopencm3.git

Now I navigate to the folder `examples/stm32/f4/stm32f429i-discovery/lcd-dma/` where I find a `Makefile`. So I issue the command `make` in that folder, and get the following output:
```sh
C:\Users\Kristof\opencm3\examples\stm32\f4\stm32f429i-discovery\lcd-dma>make
File not found - *.[ch]
process_begin: CreateProcess(NULL, #printf "  CC      sdram.c\n", ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [../../../../rules.mk:207: sdram.o] Error 2
```

I navigate to another example project, and run `make` again:
```sh
C:\Users\Gebruiker\opencm3\examples\stm32\f4\stm32f429i-discovery\lcd-serial>make
File not found - *.[ch]
process_begin: CreateProcess(NULL, #printf "  CC      sdram.c\n", ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [../../../../rules.mk:207: sdram.o] Error 2
```

I am not familiar with the codebase from `libopencm3` so it's really hard for me to debug this problem. Can anyone please help? I'm willing to make a donation to compensate for your time.

karlp, 2020-04-06:
Did you try running make in the top level directory first? as on the readme? https://github.com/libopencm3/libopencm3-examples#usage

kristofmulier, 2020-04-08:
Hi @karlp ,
Thank you for the tip. I've tried to run `make` in the toplevel directory, but I get the following error:

```sh
C:\Users\Kristof\opencm3>make
! was unexpected at this time.
make: *** [Makefile:57: lib] Error 255
```
Could it be that the makefile only works on Linux? I'm working on Windows 10.

karlp, 2020-04-09:
Honestly, I've no idea, I've been told that people have been building this on windows happily, and we've taken patches fixing various issues, so it's meant to be working.  Your particular errors don't look like the normal windows errors I've seen.  Do you have python installed?

kristofmulier, 2020-04-09:
Hi @karlp ,
Yes I've got Python installed: `Python 3.8.1`

karlp, 2020-04-09:
You might try using msys2 to get a sane build environment, but beyond that, you're going to have to find someone else who uses windows.  There might be some people on the chat channel #libopencm3 on irc.freenode.org.

kristofmulier, 2020-04-09:
Hi @karlp ,
Thank you for your quick reply. Unfortunately, I can't reach the page `irc.freenode.org`.

I hope the build-issue on Windows gets fixed. The `libopencm3` looks like an awesome initiative. It would be sad if all Windows users get excluded from this project (or if they have to apply all sorts of workarounds to get it working).

ChuckM, 2020-04-10:
A couple of things, the first is that the reason that you had an empty libopencm3 is because libopencm3 is a _submodule_ of the libopencm3-examples repository. You can get "both at once" using the command

`git clone --recursive git@github.com:/libopencm3/libopencm3-examples`

However, if you don't use the recursive flag, it is possible to move to the libopencm3-examples directory and type `git submodule init` followed by `git submodule update` which will bring you the version of libopencm3 that the examples repository is expecting.

The second thing is that you do not really describe you Windows setup. As a result there can be several issues there which are less understood.

If you are using Windows System Linux (WSL) (which is my recommended way of working this) then the instructions for Linux work fine. You may want to `cd  /mnt/c/Users/<your-user-name>/` which will connect you to the directory where your files on windows are stored by default, that lets you edit things with your windows tools and type make in the WSL window. WSL is part of Windows 10 and can be installed by going to the Windows Store and looking for Ubuntu from Microsoft.

If you are using git bash then the rules are similar except that it is important that none of the directories in your path have a space (" ") in them. That is common on windows when people have user names like "Joe User" their home directory is `C:\Users\Joe User\` and git and bash and everything else has a really hard time with that space character. This is not a recommended way to develop.

Thirdly (bonus perhaps?) there is a tutorial for building the Black Magic probe software on Windows here: [BMP on Windows](http://www.sidprice.com/2020/03/24/building-blackmagic-probe-on-windows/)  which may or may not be helpful. The BMP uses libopencm3 as part of its firmware.

I would suggest you start again, first with the `git clone --recursive` step, then cd to the libopencm3-examples directory and type `make`. Capture the output and report back here if it doesn't build for you.

kristofmulier, 2020-04-10:
Hi @ChuckM,
Thank you very very much for your elaborate answer. This is very helpful ^_^

<h3>1. Git clone</h3>
I tried the recursive cloning, but didn't get that to work at first due to a "permission denied" error. However, the recursive cloning worked this way:
```sh
git clone --recursive https://github.com/libopencm3/libopencm3-examples.git
```

<h3>2. Build</h3>
Now I `cd` into the `libopencm3-examples` directory and type `make`:

```sh
C:\Users\Kristof\projects\libopencm3-examples>make
! was unexpected at this time.
make: *** [Makefile:57: lib] Error 255
```
As you can see, there are no spaces in my path.

<h3>3. System details</h3>
Here are some more details about my Windows computer:

1. I've got Windows 10 Pro, 64-bit.

2. I'm not using Windows System Linux (WSL).

3. I'm not using git bash.

4. I'm using the good old Windows Command Prompt.

5. I've got GNU Make 4.2.1 installed:
```sh
C:\Users\Gebruiker\beetle_projects\libopencm3-examples>make --version
    GNU Make 4.2.1
    Built for Windows32
    Copyright (C) 1988-2016 Free Software Foundation, Inc.
    License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
    This is free software: you are free to change and redistribute it.
    There is NO WARRANTY, to the extent permitted by law.
```

6. I've got `arm-none-eabi` version 9.2.1 installed:
```sh
    C:\Users\Gebruiker\beetle_projects\libopencm3-examples>arm-none-eabi-gcc --version
    arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1
    20191025 (release) [ARM/arm-9-branch revision 277599]
    Copyright (C) 2019 Free Software Foundation, Inc.
    This is free software; see the source for copying conditions.  There is NO
    warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Once again a big thank you for your help ^_^

karlp, 2020-04-10:
The problem is almost definitely that your command environment doesn't know how to run the python script irq2nvic.  (The first line of which is a #!)

If you want to work in raw cmd.exe this is going to be up to you to work out.  If you can make changes that make it work, I'm happy to try and get them integrated, but fair warning, you are very much choosing the path less travelled.

ChuckM, 2020-04-10:
HI @kristofmulier 

As @karlp karl points out, you are in uncharted waters. I would say that it is uncharted because it is, perhaps, the most difficult way I can imagine to use tools developed on Linux, in Windows 10. So it is not something people have done before and so there is no experience on where the existing tools break.

I really really encourage you to go the WSL route. Microsoft put a lot of effort and time into developing WSL for Windows for exactly this reason. It not only makes running Linux tools on Windows pretty seamless, it also avoids the challenges that arise when the Linux tools get updated and you need to update your environment. Everything is in one place. It is fully supported by Microsoft and its free for anyone who has Windows 10, just install it from the store, and since it uses your windows kernel all your drivers work too. If you want to run Linux window based applications (rather than say terminal based) then you will also need X410 (also available in the store) that provides an X window server for Linux applications.

The second best way to use the system on Windows would be to develop a project file for Visual Studio Code. VSC is free as well and supported by Microsoft. In the project scripts you can mimic everything that make does and invoke python to generate the irq2nvic files. There are people who have developed support for debuggers such as the ctxLink, the BMP, and OpenOCD for VSC as well which gives you the IDE experience of seeing your source code while you run it. 

Finally, I completely understand the desire to take the road less traveled and to master its challenges. If that is what you are up for we can give you moral support and cheering on but we don't have a lot of Windows experts in the community who understand CMD or PowerShell well enough to guide you.

kristofmulier, 2020-04-11:
Hi @karlp and @ChuckM ,
It works perfectly with the WSL (Windows Subsystem for Linux). Thank you very much for your elaborate help.

I've documented each and every step in a `windows_readme.md` file, which can be useful for other Windows-folks to get started with `libopencm3`. You can download the file from this link:
https://embeetle.com/downloads/misc/windows_readme.md

*Note: I'm part of a startup building a new IDE for microcontrollers: "Embeetle". I've put the file `windows_readme.md` on our server, so you can easily download it from the link above.*

To show my appreciation for your help (and for `libopencm3` in general), I'd like to give a donation. Do you have a donate button somewhere?

Kind regards,
Kristof Mulier

ChuckM, 2020-04-11:
Thanks @kristofmulier , @karlp I read through the readme and it would be a good addition to the libopencm3 repo. 

@kristofmulier, if you're so motivated, the steps for creating a PR for adding this would be to do the following steps:
  * Clone the libopencm3 repo on your own github
  * Open up WSL and clone it (you can use the home directory of WSL, no need to go to /mnt/c)
  * NOTE: Cloning just libopencm3 does NOT need to be recursive as it has no sub-modules.
  * Now cd to libopencm3 and type `git checkout -b windows-readme`, this will make a branch from master that exists for holding the new readme.
  * Put the readme file at the top level (the libopencm3 directory where the other readme files are)
  * Then do a `git add windows_readme.md` followed by `git commit`. That will open an editor for your commit message and put "Add readme file for windows 10 users" as the commit message.
  * Now do a `git push -u origin windows-readme` and that will push your new branch back to github.
  * Finally, on github, select your branch and then click the button on the right that says "New Pull Request"
  * I will fill out the details from your commit message, just click "Submit"
  * Poof it will show up in @karlp's queue and when he gets a chance it will show up in the release.

I know , its a lot of steps but once you've done it a few times it gets to be second nature.

kristofmulier, 2020-04-13:
Hi @ChuckM ,
Thank you very much for the elaborate help! I've followed your steps one-by-one but got stuck at the `git push` step:

```sh
C:\Users\Kristof\projects\libopencm3>git push -u origin windows-readme
remote: Permission to libopencm3/libopencm3.git denied to kristofmulier.
fatal: unable to access 'https://github.com/libopencm3/libopencm3.git/': The requested URL returned error: 403
```

I also want to apologise for my late response. I've got an infection that flared up this weekend, and was unable to do any work.

Kind greetings,
Kristof

stoyan-shopov, 2020-11-05:
Sorry, I saw this thread too late.
I am also working in Windows 10, building libopencm3 there with MSYS2 works without anys issues for me.
Hope this helps.

Microsparky, 2020-11-06:
I have everything working in WSL except pyocd which is running in Windows because USB devices are not supported in WSL. 

I have been trying to get set up in Windows and had similar issues to the above running in powershell. It wasn't clear to me either that msys shell must be used as the build environment. 

My errors are shown abridged below, I assumed this meant the some portion of the Makefile system wasn't portable to Windows:
```
PS E:\...\microsparky-blink\src> make
process_begin: CreateProcess(NULL, env python E:\...\microsparky-blink\libopencm3\scripts\genlink.py ../libopencm3/ld/devices.data stm32f303ret6 FAMILY, ...) failed.
process_begin: CreateProcess(NULL, env python E:\...\microsparky-blink\libopencm3\scripts\genlink.py ../libopencm3/ld/devices.data stm32f303ret6 SUBFAMILY, ...) failed.
process_begin: CreateProcess(NULL, env python E:\...\microsparky-blink\libopencm3\scripts\genlink.py ../libopencm3/ld/devices.data stm32f303ret6 CPU, ...) failed.
process_begin: CreateProcess(NULL, env python E:\...\microsparky-blink\libopencm3\scripts\genlink.py ../libopencm3/ld/devices.data stm32f303ret6 FPU, ...) failed.
process_begin: CreateProcess(NULL, env python E:\...\microsparky-blink\libopencm3\scripts\genlink.py ../libopencm3/ld/devices.data stm32f303ret6 CPPFLAGS, ...) failed.
../libopencm3/mk/genlink-config.mk:47: No match for the FPU flags
../libopencm3/mk/genlink-config.mk:52: stm32f303ret6 not found in ../libopencm3/ld/devices.data
../libopencm3/mk/genlink-config.mk:63: ../libopencm3/lib/libopencm3_.a library variant for the selected device does not exist.
make: *** No rule to make target `/usr/lib/gcc/arm-none-eabi/6.3.1/include/stdint.h', needed by `bin/main.o'.  Stop.
```

@stoyan-shopov could you give a brief description of the MSYS2 setup process?

Thanks!

stoyan-shopov, 2020-11-09:
@Microsparky Sure. I have simply used the msys2 installer from the official page:
https://www.msys2.org/

I don't remember the installation procedure, it should be straightforward, but there is this thing that confuses me - I have msys2 installed in `C:\msys64`, and there are 3 shells in that directory: `msys2.exe`, `mingw32.exe`, and `mingw64.exe`. My understanding is that these shells start three different build environments - for `msys2`, `mingw32`, and `mingw64`. Don't ask me what the exact difference is, I don't know, and that confuses me. One thing that differs for sure is the searched include and libraries paths.

Libopencm3 builds fine in all three shells, it should not matter, because it uses the `arm-none-eabi-gcc` cross compiler toolchain. One thing to note is that the default windows search path is not available by default in these shells, so the `arm-none-eabi-gcc` compiler is not visible in those shells out of the box. To make the default windows search path available, one needs to edit file `msys2.ini`, or `mingw32.ini`, or `mingw64.ini`, depending on the shell used.

I also use `msys2` to build the `blackmagic` firmware. It also builds fine in all three shells. However, for some reason, if you want to build `blackmagic-native`, that is, to make an executable that will run on windows, you *MUST* use the `mingw64` shell, otherwise you get compilation errors.

I have also successfully compiled some other programs, like gdb and libusb, I think the shell that is used for the build environment also matters, and my guess is `mingw64` should be used as well.

The `packman` package manager program is used for handling packages in msys2, it is described on the official page.

Other than that, I don't have much experience. I have not used WSL, and pyocd, so I can't be of help there.

Hope that helps.

Microsparky, 2020-11-09:
Thanks, this thread has been really helpful!

smithps, 2022-03-03:
It's a bit late commenting on this now, but I stumbled across this (and a hacky solution) whilst working on another embedded project. I add this note for others who might stumble across this posting.

The OP states he is running Make under **Windows**....

The issue is that the makefile is written for GNU make and expects to be running under bash shell (or similar) and tries to issue several commands such as pwd, printf, rm etc... the hacky fix is to create batch files somewhere in the path that provide the commands rm, printf, etc.

The real fix is to run make under **WSL** (Windows Subsystem for Linux), or Windowsify the commands that it doesn't understand.