libopencm3/libopencm3 #1204: libopencm3 with STM32CubeIDE

imrahjel, 2020-04-08:
Hello,

I tried to use libopencm3 within STM32CubeIDE but found no obvious way to make it work.

my setup is the following:
- STM32CubeIDE
- libopencm3 for which I did "**$ make -C libopencm3 #**" as stated in the documentation
- libopencm3 template
- libopencm3 example (USB MIDI)

I started with an attempt based on libopencm3 template in which I pasted the code of an example (USB MIDI).

I found no proper way to import into STM32CubeIDE workspace. Each time I ask the IDE to build, it fails. Also, the build target (... STM32...) is disabled. Reading the Internet (not all of it of course) I found some hints that the issue might be around the linker. I tried many things in the project properties, but definitely without success.

If I create a new project from the IDE targeting my board, I have no issue to build and debug, but then I don't have libopencm3.

So, my question is : can someone explain to the noob I am how to proceed in order to get this duo working smoothly ?

many thanks,
Imrahjel.

imrahjel, 2020-04-09:
Hi,
Well, after playing with commandline and verbose options, I managed to make it work.
I share in case of.

The problem was about the 'ld' called from 'gcc' during compile phase. 'ld' was not able to find the library I was referring to.

In the project properties, go to "C/C++ General" then "Paths and Symbols".
- in the "library paths" tab, you should point to where the library is, without ending "/"
- in the "libraries" tab, you should add the name of it, without ending dot and extension, and also without starting "lib" in the name of it.

the issue was that my library was named libazerty.a so "ld" was adding "lib" before the library name, thus trying to find "liblibazerty.a"

et voilà,
Imrahjel

karlp, 2020-04-09:
great, glad you got it working.

closed 2020-04-09 by karlp