libopencm3/libopencm3 #1219: USB documentation issues

gkeeth, 2020-05-17:
I noticed several documentation issues for USB:

1. Documentation for several subclasses is missing from the online documentation, even though the files themselves are doxygen-commented: audio (usb/audio.h), CDC (usb/cdc.h), DFU (usb/dfu.h), and MIDI (usb/midi.h). 
HID and MSC documentation is present, however. 
I don't really know anything about doxygen but I think the solution might be to add a stub source file for the missing docs in lib/usb/, which was already done for HID (see lib/usb/usb_hid.c, which is noted as a stub for doxygen)

2. The comment on ```struct usb_interface``` (usb/usbstd.h) says "you should never need to instance this struct in user code!" As far as I can tell this comment is false. All of the examples instance it for each interface (see [here](https://github.com/libopencm3/libopencm3-examples/blob/9b7c2c3dbf31a92a1d970910f58ff612190971da/examples/stm32/f1/stm32-h107/usb_simple/usb_simple.c#L53) for example), and my code doesn't work if I don't instance the struct and reference it in the usb_config_descriptor. Perhaps there's a way to make it work, but none of the examples use it? Or maybe I'm dense?

karlp, 2020-05-18:
Agreed, documentation can always be improved, and is a perpetually ongoing task. Do you want a particular issue kept open for these sub classes though?  For point 2, I believe the intent is that yes, you have it (const) in your descriptor, but you never need an instance of it in your user code that _does things_

gkeeth, 2020-05-18:
My point for the documentation was that there is a page for "USB HID Type Definitions" but no page for "USB CRC Type Definitions". Audio, DFU, and MIDI are also missing. The headers themselves are already documented in the correct format, as far as I can tell, so I think the work has already been done, they just don't show up in the docs. 

I would expect the "missing" pages to show up here:
[image]

The documentation is quite sparse, but improving that is a separate issue and I understand that's difficult. I think solving the issue I'm pointing out should be trivial :).

Regarding my second point: I see what you mean, but that comment is also true for many of the other USB structures, no?

gkeeth, 2020-05-18:
To elaborate, I think the solution might be just to repeat df15b263d287718a0bb1d5abecc208b2aa3e5b7b for the "missing" headers (Audio, CDC, DFU, MIDI)

karlp, 2020-05-19:
I guess what I'm trying to say is, I don't see a lot of value in leaving the ticket open.  You are extremely welcome to provide some doxygen fixes, I agree there's missing docs on some of the usb classes.  I'm not going to spend time on it personally though, so I'm not sure we really need a special  issue for it?

gkeeth, 2020-05-19:
Fair enough. I'll see if I can figure out how to fix it and put together a PR.

gkeeth, 2020-05-23:
I'm trying to make a PR to address 1) above, but I can't get my local documentation build to match the online documentation. The online docs include a lot of content that isn't included in my results. Here's what I get: 
[image]
versus online docs:
[image]

I'm running ```make doc``` from ```libopencm3/doc```, as described in the readme. I'm having a hard time tracing through the doxygen build system, but it looks like the appropriate source files & headers are included in the the ```doxy.sourcelist``` files. 

Any pointers? What's done for the online docs that's different from what I'm doing?

karlp, 2020-05-24:
have you done a clean make from the root directory first? it looks at some of the compiled and generated output to work out other files to include.

gkeeth, 2020-05-25:
I did do a clean make first, but that wasn't the issue.

The issue was that I didn't have ```realpath``` available on OSX, which meant that malformed paths were being printed into the doxy.sourcelist files, and therefore doxygen wasn't able to find any of the files grepped out of the .d dependency files. Solved by installing coreutils. 

Sorry for the noise,

karlp, 2020-05-25:
no problem, that's a dependency we didn't know we had, I'll try and get the docs updated to cover it!

gkeeth, 2020-05-25:
See #1221 to address point 1) above.

I'm also interested in fleshing out some of the documentation around USB. I see that there has been some effort in PRs #550, #924, and very recently #1220.

Some questions: 
1. Is it okay to leverage the work done in those unmerged PRs (#550 and #924 ), i.e. is it fair for me to use the comments those fine people already wrote rather than rewriting them all? What would be the best way for me to credit them?
2. If I put together a **focused** PR that only improved documentation, would it stand a good chance of being merged? To be blunt, I'm not very interested in doing the work if it's unlikely to get merged. It looks like #550 had some open questions and was never reviewed, and #924 was reviewed but not merged, although it wasn't strictly focused on documentation and maybe that made it harder to merge the documentation parts.