libopencm3/libopencm3 #1230: adc_set_regular_sequence() is broken

rma-x, 2020-07-01:
The `adc_set_regular_sequence()` functions in adc_common_v1.c and adc_common_v2_multi.c ccontain several expresions that use `&` instead of `&&` and thus they don't work for sequences of more than four (v2) or six (v1) channels.
Additionally I think the whole functions should be refactored to be more efficient.
I plan to provide a patch.

karlp, 2020-07-02:
cute :)  WHile you're totally correct that it should be &&, because all the tests are bools, the & actually does work anyway, (but, by accident)  If you'd actually tried this you would have seen that the sequences do actually get set correctly.

rma-x, 2020-07-02:
Yes, it occurred to me afterwards that it might actually work, but only by relying on unguaranteed behavior of comparison operators.

commit 24bef9c4 (2020-07-03): stm32:adc: Change bitwise AND to logical AND

The original bitwise AND was _functionally_ correct because all operands
were booleans, but it was very poor at conveying the intent.

Fixes #1230
closed 2020-07-03 by karlp