libopencm3/libopencm3 #1236: Reset of LPTIM

AndreiCherniaev, 2020-07-19:
There is LPTIM timers. You can not set counter register of it, but you can reset counter value.

There is no special function for it. You need write 
(&hlptim1)->Instance->CR |= LPTIM_CR_COUNTRST;
to reset CNT register of LPTIM1. 

My suggestion is add macro #define __HAL_LPTIM_RESET_COUNTER(__HANDLE__)  ((__HANDLE__)->Instance->CR |= LPTIM_CR_COUNTRST)
Example of use:
__HAL_LPTIM_RESET_COUNTER(&hlptim1);

More information about LPTIM look at [RM0434 ](https://www.st.com/en/microcontrollers-microprocessors/stm32wb-series.html#resource)

karlp, 2020-07-31:
The apis you're referring to are not libopencm3, are you sure you filed this in the right project?

karlp, 2020-10-07:
if you really do believe this is a relevant bug report for this project, please re-open this with more information.

closed 2020-10-07 by karlp