libopencm3/libopencm3 #1252: Integrating GNU Build ID into the build

krish2487, 2020-09-24:
Hello,
I have a rather unusual request that someone might help me with.
I would like to add the gnu build ID into the binary that I am compiling.
The normal way to do that is to add the `-Wl,--build-id` flag to the linker flags in the rules.mk file at the top level directory.
However, doing so results in a half a gigabyte bin file.
The fix to this is to add the following section to the generated linker script, after the .text section
```
.gnu_build_id :
{
  PROVIDE(g_note_build_id = .);
  *(.note.gnu.build-id)
} > rom 
```

Now the problem is that the generated linker script is deleted and regenerated every time we run clean and build respectively. So everytime, I should remember not to run the clean command after making the changes to the linker script, or to make the changes and rerun the build.

One hack I came up with is to comment the `GENERATED_BINS += $(LDSCRIPT)` in the following section of the rules.mk file

```
# error if not using linker script generator
ifeq (,$(DEVICE))
$(LDSCRIPT):
ifeq (,$(wildcard $(LDSCRIPT)))
    $(error Unable to find specified linker script: $(LDSCRIPT))
endif
else
# if linker script generator was used, make sure it's cleaned.
GENERATED_BINS += $(LDSCRIPT)
endif
```

but it looks like a rather ugly hack and I am not sure as to what it might break elsewhere.
I was wondering if the kind people here can suggest how to implement the gnu build id into the build system without touching any of the automatons.

Any help / suggestions are welcome and appreciated. :-)

Thank you and have a nice day!!

karlp, 2020-09-24:
try just changing the linker script generator to generate what you need instead.

krish2487, 2020-09-24:
@karlp  Thanks for the input.
I am not too sure where and how to make those changes.
On the other hand, your suggestion did spark another idea.
I added the same changes I mentioned to the linker script, to the generic linker template file, like so.
under ld/linker.ld.S
```
/* Define sections. */
SECTIONS
{
	.text : {
		*(.vectors)	/* Vector table */
		*(.text*)	/* Program code */
		. = ALIGN(4);
		*(.rodata*)	/* Read-only data */
		. = ALIGN(4);
	} >rom
	.gnu_build_id :
	{
	  PROVIDE(g_note_build_id = .);
	  *(.note.gnu.build-id)
	} > rom 
	/* C++ Static constructors/destructors, also used for
	 * __attribute__((constructor)) and the likes.
	 */
	.preinit_array : {
		. = ALIGN(4);
		__preinit_array_start = .;
		KEEP (*(.preinit_array))
		__preinit_array_end = .;
	} >rom

```

And this seems to have the same effect as prepopulating the generated linker with the build id section. 
:-)
I am not sure if this is the correct way, but it might help someone later. 
I am closing the issue for now. 
I apologize for wasting your time.

closed 2020-09-24 by krish2487

karlp, 2020-09-24:
yes, that's what I mean, change what gets generated, don't try and modify it after it's been generated :)  glad it worked out for you