libopencm3/libopencm3 #1263: Building libopencm3 for a specific target

krish2487, 2020-10-29:
Hello,
I know this look like a rather silly issue.
A while ago, I was able to compile libopencm3 for a specific stm32 family (F1 to be precise)  to keep the footprint low. 
I do not remember for the life of me, what I make file target I gave it.

Can anyone pitch in and help me with this?

To be clear, I am using the libopencm3 template and would like to build the libopencm3 library - but only for the STM32 F1 family. not the rest. I know it can be done and I have done it before from the makefile in the libopencm3 top level directory.. but I do not remember how..

krish2487, 2020-10-29:
Never mind,

As is the case, I figured out how to do so a couple of minutes after I posted the question.
🤦 

For anyone else who might want to do the same, this is how.

` make $TARGETS=stm32/f1`

I am closing the issue now.

karlp, 2020-10-29:
No, it' sok, we should add that to the front readme page under "building" seeing as we support it.

karlp, 2021-12-01:
fixed via the linked issue being merged

closed 2021-12-01 by karlp