libopencm3/libopencm3 #1265: Either avoid using `python` (without version) or suggest using a virtualenv.

sumpfralle, 2020-11-06:
I posted the following text one week ago as [comment for #1075](#1075#718045397). But that issue was already closed for some time, thus my comment was probably more or less invisible. Feel free to close this new issue and re-open the old one, if you think, that the discussion should continue there instead of here.

Summary: I would like to re-open the discussion regarding the most suitable python-shebang in the four python scripts used in libopencm3. The previous discussion (#1075) seems to contain a few misinterpretations of the relevant [PEP-0394](https://www.python.org/dev/peps/pep-0394/), which I would like to clarify below in order to smoothen the usage experience of libopencm3.

> According to https://www.python.org/dev/peps/pep-0394/ we should keep using "python" if we want to support both 2 and 3, and just make sure our script works with both, which I believe it does, but feel free to correct me on that.

Indeed [PEP-394](https://www.python.org/dev/peps/pep-0394/#for-python-script-publishers) allows `#!/usr/bin/env python`, but it should *only be used in a virtualenv*:

> For scripts that are only expected to be run in an activated virtual environment, shebang lines can be written as #!/usr/bin/env python, as this instructs the script to respect the active virtual environment.

I guess, that we do not really expect the users of libopencm3 to run `make` (for compiling libopencm3) in a python virtualenv, or?

PEP-394 further says:

> Scripts targeting both “old systems” and systems without the default python command need to make a compromise and document this situation. Avoiding shebangs (via the console_scripts Entry Points ([9]) or similar means) is the recommended workaround for this problem.

Thus the minimum effort recommended by PEP-394 would be to document, that `make` should be typed in a Python virtualenv. Other approaches could be a new pre-compilation step converting the input python scripts into *locally usable* python scripts with a proper (detected) shebang. But this feels unreasonably cumbersome.

> If fedora is breaking the PEP by simply having no python executable, then, respectfully, I feel that's a fedora bug. (And thanks for hitting it before me, I'm still on fed29)

No, I think, Fedora's removal of `python` does not break PEP-394, but follows its wording perfectly.
See one of the [proposed options](https://www.python.org/dev/peps/pep-0394/#for-python-runtime-distributors):

> Distributors may choose to set the behavior of the python command as follows:
> * ...
> * not provide python command,
> * ...

At least Fedora and Debian (and thus probably all Debian-derivates like Ubuntu, Mint, ...) will *not* ship `/usr/bin/python` at least for one release, in order to give local administrators a clear indication, that they should take a look at certain (old) scripts and check, whether they are ready for switching to another language (python3). *Maybe* these distributions will start providing `/usr/bin/python` in a later release again (after everyone noticed during the previous upgrade, that their local scripts need attention). But this will depend on whether anyone will be interested in `/usr/bin/python` at this point in the future. Since every PEP is recommending to use `python3` (instead of `python`) for years, there may be no need for `/usr/bin/python` in a few years.

Now we are close to the end of 2020:
* python3 is available in every major distribution for many, many years (since ~2010)
    * RHEL5 is probably the last distribution, shipping only python2 (even though there are probably ways to install python3)
    * RHEL5's *Extended Lifecycle Support* will end this month (November 2020); its general *maintenance* period expired three years ago
* python2 is being removed (or was removed some time ago) from major distributions due to the end-of-life of Python2 (January 2020)

Conclusion:
* `python` is not available on some current distributions and will be missing from *most* distributions in the next years
* `python3` is available in *all* currently supported distributions
    * at least after November 2020 (due to the end-of-extended-life for RHEL5, see above)
* sticking to the `python` approach for shebangs would require us to teach users to use a Python virtualenv when running `make` or to provide some fancy detection of the python interpreter in libopencm3's Makefiles

Thus I would recommend to switch the shebang of the four related scripts below `scripts/` from `#!/usr/bin/env python` to `#!/usr/bin/env python3`. This will ease the life of the majority of our current and future users, while causing suffering only for users of pre-2010 distributions, which are technically really out-of-date and unsupported.

The future (python3) is here - let us welcome it! :)

btw: my pre-make workaround for projects using `libopencm3` is the following:
```shell
grep -rl --null "/env python$" build/libopencm3/scripts/ | xargs -0 -r sed -i '1s#/env python$#/env python3#'
```
I would really like to get rid of this workaround.

ret394, 2020-11-24:
> The future (python3) is here - let us welcome it! :)

Yes, welcome it in the future.

---

Use Python 3 if there's a whole lotta benefits.

Anyway, if you see there is no benefit to switching to python3.
Python 2 is the future.

Python2 will always be available on non Windows and Windows systems. With the name python2 or python2.7. If it is not call your sysadmin and tell them to 'just do it', if they refuse. Well, if sysadmin is what my kids can do after years of training. I use scripts for that, a one liner.

you're welcome, as always. all the best. 👍
don't say nasty stuff or do rude stuff to folks on Github or other sites.

[ret394](https://github.com/ret394)

sumpfralle, 2020-11-24:
Thank you for your comment!

> Python2 will always be available on non Windows and Windows systems.

Of course, you will be able to install it manually somehow, but *every* distribution will need to phase it out slowly, since Python2 is end-of-life for almost a year now. I detailed this in the description of this issue.

Summary of the current state:

* Starting from the first of December 2020 *every* maintained distribution release includes python3.
* The libopencm3 scripts in question are compatible with python3.

Conclusion: the libopencm3 build process would *just work* everywhere, if the shebang would change to `python3`.

In case someone disagrees: please state an explicit benefit of sticking to `python` (instead of `python3`). Thank you!

ret394, 2020-11-24:
> ...

Private message me on Gitter. And state the payment amount. I can't keep doing this for free, it's worth millions of USD.

2 secs due to page load
poetic ret394

commit 5968b8a8 (2020-11-28): Update python references to python3

Python2 is end-of-life [1] since the 1st of January 2020.

Some distributions (most notably: Debian and its derivatives) will stop
providing a `python` executable in order to encourage users to specify
the interpreter language of local scripts explicitly.  Users of such
environments will be forced to work around this in one of these ways:

* create a virtual environment or
* manipulate the shebangs of the scripts or
* install the python2 package (as long as it is provided by distributions)

All currently maintained distribution releases provide python3.
In  [... cut]
closed 2020-12-01 by karlp