libopencm3/libopencm3 #1267: usb support for STM32F7

engelant, 2020-11-18:
Has anyone looked into the differences of the USB cores between STM32F4 and STM32F7?
Are they like completely different or mostly similar? Is there any doc available on that topic?
Has there any work been done in the direction of supporting STM32F7 USB?

manuelbl, 2020-11-18:
The USB OTG cores are more or less the same for all STM32 chips. They can be divided into those with core ID 0x1000, 0x2000 and 0x3000. Anyhow, libopencm3 has currently several issues with USB and there are pending PRs for most of them.

Create a branch and apply these PRs:

- #1256 Disable VBUS sensing
- #1258 Fix usb_dwc_common.c endpoint initialization
- #1259 Fix usb_dwc_common.c late IN usb interrupt flag clearing
- #1261 USB Core 0x2000 fix
- #1264 USB OTG support for STM32F7 series

Then it will work for your STM32F7.

engelant, 2020-11-18:
I wasn't able to find any of this with google...
However, it's nice to know this is been worked on.

From what I red it seems that the current holdback is more Jenkins than code related.

I think as of for right now I'll just keep an eye on this development and hope it gets merged into master eventually.

closed 2020-11-18 by engelant

karlp, 2020-11-20:
@engelant for many users, you only need https://github.com/libopencm3/libopencm3/pull/958 and it "just works" but not all boards and not all users have success with just that.

manuelbl, 2020-11-20:
@karlp I doubt that #958 is sufficient. Additionally, you need to take care of VBUS sensing (can be done in hardware or using #1256) and if the USB core has ID 0x2000 it will exhibit further problem unless #1256 (and probably also #1258 and #1259) are applied.

It seems that libopencm3 hinges solely on you and that you can only dedicate limited time to you. The project is stuck that way. What changes need to be made to revive the project and bring speed and life back?

karlp, 2020-11-20:
not everyone needs vbus sensing changes, at all.  the people who submitted 958, and others at the time had all things working for them on their hardware, including all gadget0 tests.

manuelbl, 2020-11-20:
Aren't you interested in better supporting the 99% case? Currently, libopencm3 turns on VBUS sensing. If you are not aware of it, USB won't work. And it's not documented, it's just an unnecessary pitfall. Furthermore, the libopencm3 USB stack is limited to USB device and doesn't support the OTG host mode. In that context VBUS sensing has no real use. So at the moment, the VBUS configuration is very close to a bug.