libopencm3/libopencm3 #1270: better gpio_toggle()

nerdralph, 2020-12-04:
https://github.com/libopencm3/libopencm3/blob/f2a9980638a0f52368a0c76fe12ef88e9ecb07c5/lib/stm32/common/gpio_common_all.c#L87

this way saves 8 bytes:
`GPIO_ODR(gpioport) ^= gpios`

manuelbl, 2020-12-04:
There's a major difference between the current implementation and your proposal. Think about what happens if an interrupt occurs after the register has been read and before it's written...

The shorter version isn't interrupt safe as it potentially writes old values to the register.

nerdralph, 2020-12-04:
The function already isn't atomic, and it is clearly documented as such:
"The gpio_toggle() operation is the only GPIO port operation which is not
atomic."

nerdralph, 2020-12-04:
However I do see that the current version, while not atomic, will not modify any of the bits in gpios that are not set.  So as long as interrupt code doesn't modify any of the same bits, gpio_toggle() acts as if it were atomic.  I think it would be useful for this to be documented.

So without sacrificing the pseudo-atomic behavior, there's still room for a small optimization by eliminating the shift and or operations to save 4 bytes:
`uint32_t port = GPIO_ODR(gpioport);
GPIO_BSRR(gpioport) = (~port & gpios);
GPIO_BRR(gpioport) = (port & gpios);
`

manuelbl, 2020-12-04:
I just wanted to point out the same regarding the bits not specified by `gpios`.

The proposed optimization looks ok. I'm surprised it's shorter.

nerdralph, 2020-12-04:
> 
> 
> I just wanted to point out the same regarding the bits not specified by `gpios`.
> 
> The proposed optimization looks ok. I'm surprised it's shorter.

Thumb-2 drops the ability to use the barrel shifter on every instruction, so the 16-bit shift isn't free.  There is an extra str instruction, but since BSRR and BRR are close (16 bytes), it's just using a different immediate offset.

karlp, 2021-01-02:
If you have specific code changes, please make them as pull requests, not as issues.

jcw, 2022-01-30:
If cycle-shaving is the goal, then this code can be tweaked:

```
void gpio_toggle(uint32_t gpioport, uint16_t gpios)
{
	uint32_t port = GPIO_ODR(gpioport);
	GPIO_BSRR(gpioport) = ((port & gpios) << 16) | (~port & gpios);
}
```

... to the following:

```
void gpio_toggle(uint32_t gpioport, uint16_t gpios)
{
	uint32_t port = GPIO_ODR(gpioport);
	GPIO_BSRR(gpioport) = (gpios << 16) | (~port & gpios);
}
```

The reason is that the SET bits take precedence over the RESET bits (and if the code is inlined and `gpios` is a constant, then the shift can be done at compile time).

karlp, 2022-02-01:
Same comment as before, if you have specific changes, please make them as pull requests.

karlp, 2023-11-05:
again, these need to to be pull requests to be considered seriously, we'r enot having code discussions on issues.

closed 2023-11-05 by karlp