libopencm3/libopencm3 #1271: add gc-sections to makefiles

nerdralph, 2020-12-08:
The default makefile flags already include -ffunction-sections and -fdata-sections.  Adding -Wl,--gc-sections saves flash space by omitting the unused sections.  For example it saves 440B on the miniblink example when compiled for the f030.

karlp, 2020-12-08:
that's something you do when you link your application, not in the library...  The library has to have everything available for it so that people can use it, and they can gc sections in their app.  Do you mean to file this ticket on some other repository?

nerdralph, 2020-12-08:
This isn't the "root" repository for issues related to multiple of the other repositories?
I thought about opening the issue on the miniblink, examples, and templates repos, but thought it makes more sense to open a single issue.

karlp, 2020-12-10:
no, make issues in the repositories where the issues are please.
Note that saving space in the miniblink example is completely irrelevant, there's no way you should be copying that repo's makefiles for any real project, and if you do, you're completely on your own :+1: 
The libopencm3-template repo, which _is_ intended to be copied, does include gc-sections like a normal app.  https://github.com/libopencm3/libopencm3-template/blob/master/rules.mk#L87
The libopencm3-examples repo is ~EOL, but if you want to make changes there, you're free to send them as PRs I'm closing this issue on _this_ repository though.

closed 2020-12-10 by karlp