libopencm3/libopencm3 #1272: consider inlining gpio_set for stm32

nerdralph, 2020-12-09:
On the f0, l0, g0, f1, f4, and possibly all stm32 MCUs, gpio_set compiles to 2 instructions: str + bx.  The call to the function is a 32-bit bl instruction.  Inlining the function never take more code, will always be faster, and will sometimes take less code because it will reduce register pressure on r0-r3.
For those targets with BRR, the same logic applies to inlining gpio_clear as:
`GPIO_BRR(gpioport) = gpios;`

karlp, 2020-12-10:
if you're at all concerned abotu these sorts of things, _most_ of libopencm3 is not going to be what you want.  I'm strongly against fairly arbitrary scattering of inlines around the code base. (because as soon as you start, everyone wants their special functions inlined)  I'm also led to believe that LTO "just fixes" this if you do it that way.  Certainly, this sort of feature is unlikely to be worked on from an issue, at least, not by me :)

karlp, 2020-12-10:
See also https://github.com/libopencm3/libopencm3/issues/661 for past efforts in this direction.

closed 2020-12-10 by karlp