libopencm3/libopencm3 #1273: CMSIS_5 support

mttbx, 2020-12-10:
CMSIS_5 is under Apache-2.0 License. It is good for porting. Can we add support to CMSIS_5? I think this would be good for libopencm3.

karlp, 2020-12-10:
What exactly do you mean by "support to cmsis-5" ?

mttbx, 2020-12-10:
CMSIS-5 library contains many useful functions which can help to porting. We can just use CMSIS-5 instead of implementing the same functions by ourselves. And focus to other functions. The problem I encountered is that if I include both libraries (libopencm3 and CMSIS-5) into my project, there would be some redefine issue.

mttbx, 2020-12-10:
I think this would make this library better. Just use CMSIS-Core would be enough.

karlp, 2020-12-10:
yes, these days, cmsis licenses would allow us to have built it all on top of cmsis.  but doing so now is a decidedly non-trivial piece of work.  I'm not against doing it, but I'm not going to keep a generic issue around tracking it, as I'm certainly not putting in the time to do so.  If you can come up with some concept PR, we can look at it

closed 2020-12-10 by karlp